lowRISC/ibex #205: Verify debug interface

udinator, 2019-08-01:


imphil, 2019-08-08:
See also #168

arghimat, 2019-09-10:
Is the software debugging currently working for ibex? Is there a Debug Module (DM), Debug Module Interface (DMI) and Debug Transport Module (DTM) as specified by the RISCV specification already available or planned in this project? I would like to use OpenOCD with ibex but I am not sure if this project will provide the required debug infrastructure or this is not in the scope.

imphil, 2019-09-10:
You can use the debug module from PULP https://github.com/pulp-platform/riscv-dbg/ with Ibex.

udinator, 2019-10-31:
All features of Debug-Mode currently implemented have been tested, and a main issue was found with how `dret` was treated in normal execution modes. Fixes have been pushed for this bug, so will close this issue as testing is complete.

closed 2019-10-31 by udinator