lowRISC/ibex #212: UNOPTFLAT in ibex_core.cs_registers_i.mie_q

imphil, 2019-08-05:
Investigate where this Verilator warning comes from, and decide if we want to fix it.

vogelpi, 2019-11-01:
I investigated this issue and found out that `mie_q` (the interrupt enable bits) via `irq_pending` directly feed into the main clock gate in `ibex_core.sv`. This one drives the clock of the register holding `mie_q`. I see the following ways to remove that, but consider none of them optimal:

1. Add another register for `irq_pending` using the un-gated clock (similar to `core_busy_q` in `ibex_core.sv`). This adds another cycle of latency for interrupts. We probably don't want that.

2. Clock the register holding `mie_q` with the un-gated clock. This does not add another cycle of latency for interrupts, but is not really clean and sub-optimal for power.

3. Replace the latch in the main clock gate with a flip-flop. The fact that once more the a latch is involved (see also lowRISC/ibex#93) shows that Verilator really does not like latches. We could extend all `prim_clock_gating.sv`-versions (see lowRISC/ibex#213) with VERILATOR ifdef. But also this does not seem to be an appealing solution and it might have other implications.

Since the problem seems to be Verilator-specific, we might want to leave it as it is and just stick to the lint waiver. WDYT?

sajjadahmed677, 2021-01-10:
I am also having same issue. is there any other way to resolve it??

rswarbrick, 2021-01-11:
Hmm, this can be fixed by telling Verilator that the `en_latch` signal in the clock gating module is a clock gate, by using the "clock_enable" hint. Since we vendor this code, I'll push up a PR to OpenTitan with the fix, and vendor it back into Ibex when it's landed.

rswarbrick, 2021-01-11:
OpenTitan PR at https://github.com/lowRISC/opentitan/pull/4792

vogelpi, 2021-01-11:
Thanks @rswarbrick for fixing this :+1:

PR #1250 (2021-01-25): [verilator] Remove clock gating waiver and add extra warning
The `-Wwarn-IMPERFECTSCH` flag is suggested in the Verilator docs (to
help spot if we get a clock gating hint wrong). The waiver is no
longer needed because we now add the relevant hint in the
RTL (vendored in `prim_generic_clock_gating.sv` from OpenTitan).

Closes #212.
closed 2021-01-25 by rswarbrick