lowRISC/ibex #217: enabling Tracer in verilator

eroom1966, 2019-08-06:
@imphil referring to #99 
Great progress, I have managed to reproduce your results, and have all but 4 RV32I tests passing (as expected)

**Q1. Is it possible to enable instruction tracing, register dumping, memory Load/Store from the simulation run ?**

I will attempt to (manually) merge your code changes into my sandbox
Looking at your changes, I see the following

Changes to link.ld, my plan is to have a local linker script for ibex, and to have an additional section for the traphandlers, in order to define the correct memory region.

In RVTEST_CODE_BEGIN, I note you have removed the line

> 110: .align 6

**Q2. can you explain why this was removed ?**

Also I note you have commented out the code initialising the (non-existant) Supervisor registers.
I had already handled this differently by indicating the the framework whether Supervisor or User modes were present, so I think I can fix this is a way that does not affect the other architectures

**Q3. Finally, do you have a memory map of the RTL platform you have created ?**
I will bear this in mind when creating the linker script.

Thx
Lee

imphil, 2019-08-06:
> Q1. Is it possible to enable instruction tracing, register dumping, memory Load/Store from the simulation run ?

The full tracer implementation doesn't work with Verilator at the moment, unfortunately (I've opened https://github.com/lowRISC/ibex/issues/220 to get this sorted out hopefully). The easiest solution for now is to apply the following patch.

```patch
diff --git a/rtl/ibex_controller.sv b/rtl/ibex_controller.sv
index 464d6a1..d996724 100644
--- a/rtl/ibex_controller.sv
+++ b/rtl/ibex_controller.sv
@@ -134,8 +134,8 @@ module ibex_controller (
   // glitches
   always_ff @(negedge clk_i) begin
     // print warning in case of decoding errors
-    if ((ctrl_fsm_cs == DECODE) && instr_valid_i && illegal_insn_i) begin
-      $display("%t: Illegal instruction (core %0d) at PC 0x%h: 0x%h", $time, ibex_core.core_id_i,
+    if ((ctrl_fsm_cs == DECODE) && instr_valid_i) begin
+      $display("%t: Instruction (core %0d) at PC 0x%h: 0x%h", $time, ibex_core.core_id_i,
                ibex_id_stage.pc_id_i, ibex_id_stage.instr_rdata_i);
     end
   end
```

On register and memory load/stores: Do you want to get a trace of all register writes and memory accesses, or do you want to dump this information on a trigger or at a certain point in time?

Going beyond that, we could add output support to our simulation (printf() through the RISCV_IO macros). Would that help your debugging?

>  Q2. can you explain why this was removed ?

The `.align 6` prevents `.org 0x0` and `.org 0x80` to work, it gets either half-ignored or the assembler complains about moving backwards (I don't really understand why).

> Q3. Finally, do you have a memory map of the RTL platform you have created ?

Sure, see below.

| Start   | End     | Size  | Device                         |
|---------|---------|-------|--------------------------------|
| 0x0     | 0xFFFF  | 64 kB | shared instruction/data memory |
| 0x20000 | 0x203FF | 1 kB  | test utility                   |

eroom1966, 2019-08-08:
@imphil 
Hmm, maybe I misunderstood ...
I was hoping that the tracer produced a disassembly output indicating the instruction, arguments and any changed registers.

I am sure in the past we have produced an execution disassembly in the Mentor or Cadence environment - am I mistaken ?

imphil, 2019-08-08:
In Mentor and Cadence simulations you can use ibex_tracer, which gives you disassembled instructions. It just doesn't work in Verilator yet (#220).

eroom1966, 2019-08-08:
Ah, ok, I thought adding the patch addressed this - my mistake
I will look at using Mentor/Cadence

vogelpi, 2019-10-31:
The tracer is now Verilator compatible. This has been added starting with #360. Also the PR upstream riscv/riscv-compliance#69 resolving this issue has been merged. I am thus closing this issue now.

closed 2019-10-31 by vogelpi