lowRISC/ibex #220: Make ibex_tracer Verilator-compatible

imphil, 2019-08-06:
It should be possible to use a coding style for the tracer which works in Verilator, making this helpful tool available there as well.

raulbehl, 2019-09-03:
This is work in progress (I actually have the branch ready just need some style-nit fixes before raising a PR). I have used #262 as my starting point for this. Should I wait until #262 is merged and then open a pull-request?

imphil, 2019-09-03:
Feel free to open a PR right away. This allows us to start with the review right away, and you can then rebase before merging.

vogelpi, 2019-10-31:
The tracer is now Verilator compatible. This has been added starting with lowRISC/ibex#360. I am thus closing this issue now.

closed 2019-10-31 by vogelpi