lowRISC/ibex #228: DV: Test writable mtvec

imphil, 2019-08-08:
With #118 landed mtvec is now writable. We need DV to test this, and to ensure that the documented constraints (256B-aligned, only vectorized) are obeyed when reading the value back.

udinator, 2019-08-08:
@imphil, I tested writes to mtvec, and its behavior is correct, it remains 256B-aligned, and the bottom two bits always remain 2'b01.

imphil, 2019-08-09:
Thanks @udinator! Did you do a manual test, or is this already part of the DV in this repository? Please feel free to close this issue once it is covered by in-tree DV.

udinator, 2019-08-09:
The testing I did to verify `mtvec` is part of my latest merge to the `google/riscv-dv` repository, and the script in question can be found [here](https://github.com/google/riscv-dv/blob/master/scripts/gen_csr_test.py). I will be updating the `riscv-dv` repository in-tree as part of my PRs that are currently outsanding for this repo.

taoliug, 2019-08-12:
@udinator, please update the boot strap setup here
https://github.com/lowRISC/ibex/blob/master/dv/uvm/riscv_dv_extension/ibex_asm_program_gen.sv#L32

Right now all the trap/interrupt handling is hard-coded to boot_addr. It should be replaced to allow programmable MTVEC.

PR #256 (2019-08-22): [DV] Make mtvec writable, remove previous workaround
fix #228
closed 2019-08-22 by taoliug