lowRISC/ibex #229: Unused signal in decoder: data_reg_offset_o

ivanmgribeiro, 2019-08-08:
Hi,
I was looking through the ibex_decoder and found the signal ```data_reg_offset_o```, and I couldn't find anywhere where it is set apart from on the line
https://github.com/lowRISC/ibex/blob/19ffe9ac1db96dcdfb18c5fe0b3b2cf382cc179b/rtl/ibex_decoder.sv#L193
Is there a reason for this signal to exist?

Thanks,
Ivan

vogelpi, 2019-08-08:
Hi Ivan,
thanks for opening this issue. I am not 100% sure why this signal is there. It is also there in RI5CY but also not used there. I think it is a remnant of a custom instruction set extension from back in the days when PULP was using OpenRISC.

For sure it is not used and I think we should remove it.

Best regards,
Pirmin

NamelessFTW, 2019-08-12:
Hi,
I'm currently away for the next week, and I won't be able to create the PR until I'm back. If this delay is OK with you then I'll implement the changes when I'm back, otherwise feel free to implement them.

ivanmgribeiro, 2019-08-12:
Hi,
I'm currently away for the next week and I won't be able to create the PR until I'm back. If you're OK with the delay then I'll implement the changes once I'm back, otherwise feel free implement them.

vogelpi, 2019-08-20:
This got fixed in PR lowrisc/ibex#248. I am thus closing this issue.

closed 2019-08-20 by vogelpi