lowRISC/ibex #237: Performance compared to zero-riscy

mmatzev, 2019-08-13:
I'm using a modified version of the pulpino SoC and replaced zero-riscy by ibex. I use the perfbench test software from the pulpino repository and modified only the CSR access addresses to get it working. 
It is stange that ibex is taking more clock cycles for the same code as zero-riscy.  Also the number of branches is significant larger. Is there any explanation for this behavior? 
The simulated time is also slightly larger (about 10%) but in some cases the number of clock cycles are more than 50%. Therefore, I have the impression that the counters are not working as expected.

vogelpi, 2019-08-13:
Hi @mmatzev ,
thanks for opening the issue. We have reworked the performance counters some time ago and in this process also slightly changed the order of the events and how they are generated. If I remember correctly, we found out that some events were not properly generated like the instructions retired signal.

In general, the changes we have implemented should not have an impact on performance. But it is important to us to find out if there is actually a problem in the current implementation.

I think the first thing would be to check the number of cycles. Would it be possible for you to compare the values returned by the cycle counter inside Ibex with the external timer in PULPino?

Best,
Pirmin

mmatzev, 2019-08-14:
Hi,
the number of clock cycles fits to the simulation results. I compared also the code of the zero-riscy version and the ibex version. This is very similar, the only difference is in the CSRR and CSRW commands as the register addresses differ between the cores.

For the IBEX using the bench/fdctfst example from pulpino, I get:
Perf CYCLES:   16686 (CSRR 0xB00)
Perf INSN:     12250 (CSRR 0xB02)
Perf CINSN:    7246 (CSRR 0xB0A)
Perf LD_STALL: 10857 (CSRR 0xB03)
Perf JR_STALL: 6372 (CSRR 0xB04)
Perf IMISS:    4503 (CSRR 0xB05)
Perf #LD:      5181 (CSRR 0xB05) --> Read two times because of different parameter in zero-riscy
Perf #ST:      1604 (CSRR 0xB06)
Perf #JUMP:    679 (CSRR 0xB07)
Perf #BRANCH:  6658 (CSRR 0xB08)
Perf #TAKEN:   6539 (CSRR 0xB09)
Perf #RVC:     20731 (CSRR 0xB0A)

For the zero-riscy using the same SoC and same example I get:
# RX string: Perf CYCLES:   10551                                                                                                                                                                                                                                            
# RX string: Perf INSN:     7294                                                                                                                                                                                                                                             
# RX string: Perf CINSN:    2974                                                                                                                                                                                                                                             
# RX string: Perf LD_STALL: 0          --> Not implemented in zero riscy                                                                                                                                                                                                                                       
# RX string: Perf JR_STALL: 0           --> Not implemented in zero riscy                                                                                                                                                                                                                                     
# RX string: Perf IMISS:    754                                                                                                                                                                                                                                              
# RX string: Perf #LD:      830                                                                                                                                                                                                                                              
# RX string: Perf #ST:      845                                                                                                                                                                                                                                              
# RX string: Perf #JUMP:    65                                                                                                                                                                                                                                               
# RX string: Perf #BRANCH:  125                                                                                                                                                                                                                                              
# RX string: Perf #TAKEN:   109                                                                                                                                                                                                                                              
# RX string: Perf #RVC:     2974        

What looks strange:
- In IBEX. There is a large number of braches and braches taken?
- In IBEX. There is a large number of misaligned access?
- In Zero-Riscy. The number of Compressed instructions (CINSN and RVC) is the same even these should be a large number of instructions executed during printout.

What can I do to analyze this problem more deeply?

Best regards, Matthias

vogelpi, 2019-08-15:
Hi Matthias,
thanks for getting back. There is definitively something fishy going on. Already the number of loads and stores + the branches and jumps exceed the number of executed instructions on Ibex. Looking at the source code I do not find an explanation for this.

One thing that came into my mind: Do you properly reset the performance counters in Ibex? I just realized that Zero-riscy had a mechanism to reset all counters at once (by doing a CSR clear/write to address 0x79F). Ibex does not support this, every counter needs to be separately cleared.

One more question: what do you mean by RVC and CINSN?

Best,
Pirmin

mmatzev, 2019-08-15:
Dear,

I think that the counter reset and counter enable is the difference. 
In zero-riscy there was the CSR 7A1 to enable and disable all counters. Therefore, they are stopped after the bench operation. In ibex, this register does not exist and, therefore, the counters are still runing during data output. 
There is a CSR 0x320 without documentation. Should I use this register to stop the counters?

Best, Matthias

vogelpi, 2019-08-15:
Yes, you have to use the `mcountinhibit` CSR at addres 0x320. What do you mean by "without documentation"? The use of the `mcountinhibit` CSR is explained here: 
https://ibex-core.readthedocs.io/en/latest/performance_counters.html#controlling-the-counters-from-software

Do the numbers now match?

mmatzev, 2019-08-15:
Sorry, yes it is there. I had a look to https://ibex-core.readthedocs.io/en/latest/cs_registers.html and did not find it there. It would be nice to have a link to the right section.

Now, I implemented the start and stop and reset of the counters. This results in very similar values to the zero-riscy core. The difference is in the range of 10 cycles, only. 

Thanks for your support, Matthias

vogelpi, 2019-08-15:
You are welcome Matthias, it is good to know that the results are comparable.

closed 2019-08-15 by vogelpi