lowRISC/ibex #258: Signal “core idle” to users outside of the core

imphil, 2019-08-22:
We need to add a top-level signal indicating that the core is in WFI and completely idle.

It's a good indication for the power manager to know that the core is completely quiescent, and it is safe to switch off power without leaving a hanging transaction on the bus. The outside hardware world needs some kind of indication that upon WFI ibex is truly idle, and not for example secretly doing prefetches.

GregAC, 2019-09-05:
A question here is how once sleeping does the core wake-up occur?  The linked issue in the pulp repo is saying they want the signal to go low once something happens that wakes the core up.  However to do this you need a combinational path from the irq pins to the sleep signal, which I don't think is a great idea timing wise.  Perhaps a separate wakeup_needed which is just a big OR or the various signals that could trigger wakeup would work?  Though maybe it's better for this to be handled entirely externally (i.e. power manager know it needs to turn the core back when interrupts are seen).  I'm not sure combinational logic would even work if we've got the power off anyway!

The other question is if this results in an actual power down how is register state being maintained?

Also noticed something else that troubles me timing wise.  We've got the top-level IRQ being fed directly into the core clock enable.  This can maybe work but does require rather early IRQ signals.  I'll open another issue for it.

imphil, 2019-09-05:
> A question here is how once sleeping does the core wake-up occur?

Sleep is only due to WFI, so any interrupt (including the "debug interrupt") will wake up the core.

For the scope of this bug, probably passing clock_en to the outside world should be sufficient. But I agree we could go beyond that, let's just make sure we don't scope-creep too much here.

https://github.com/lowRISC/ibex/blob/36db1041605c4e0c62e5e0612f35331647674ed8/rtl/ibex_core.sv#L275-L285

> The other question is if this results in an actual power down how is register state being maintained?

I'm not sure if we really need to support power down, or if a gated clock or reduced frequency clock is sufficient. I guess "core idle" doesn't automatically imply you can power it down, but it could mean you can power down or gate other pieces in the system, e.g. bus components.

@tjaychen You originally requested support for this signal. Was your intention to use this signal to power down the core?

tjaychen, 2019-09-06:
Yeah the intent was to power it down.  It could also be clock gating, but
worst case is full power down.

Regarding the wake up path, some of the past cores I've worked with had two
clock domains to the core. One was always running and one could be gated
off based on an idle signal (either internally or externally)  The former
would be used to clock in the wake up/irq signals to wake up the rest of
the core.

Do we have something like that here? If not, I'm fine with doing it outside
also.

Regarding power down, when the system wakes up, I think it is fair to
assume the power manager will release clocks to the core, so to me it folds
into the case above.

Does that help?

Depending on what we choose to do above, we may also want to have a quick
conversation around the idle gray windows (ie, the core is actually waking
up due to irq, but the system thinks it's still safe to power off).

tjaychen, 2019-09-06:
O one thing I should have clarified for power down.

If we do power down just the core and not the rest of the system, we will
most likely isolate the output to maintain the proper state (ie in sleep),
either on the output of the core or the input to some other module,
depending on how the power domains are partitioned.

If we power down the core with the entire system, it is then the power
managers responsibility to wake up due to correct reasons and power the
core and release it from reset and once again allow execution from ROM.

Does that help?

Since ibex is a smaller core, I think most of the time it will get powered
off along the surrounding system, since we probably don't gain a ton just
power gating ibex by itself (don't quote me there, I'm just making a rough
guess, when the time comes, a more formal leakage analysis will help make
the decision)

closed 2019-09-06 by GregAC

Silabs-ArjanB, 2020-05-15:
FYI: I just filed a pull request for CV32E40P for the addition of a sleep pin (see https://github.com/openhwgroup/cv32e40p/pull/341). I believe that the way that core_sleep_o has been added inside Ibex will lead to core_sleep_o being 1 during (and just out of reset), which I don't think is what you would really want (it should onyl possibly get set in response to executing a WFI instruction).

commit bec84ca2 (2019-09-06): Add core_sleep_o to ibex interface

Signals the core is totally idle (WFI with no outstanding memory
transactions).  Fixes #258