lowRISC/ibex #261: [ibex tracer] enhancement request - runtime knob to disable tracer

sriyerg, 2019-08-23:
At chip level DV, we "stub" the ibex cpu by forcing its clk_i to 0 and taking control of the TL interface by overdriving it with our TL transactor. The advantage of doing that is choosing to 'stub' the cpu becomes a runtime config knob / setting as opposed to being a build time, which reduces overhead in terms of requiring additional parallel builds. 

In chip level dv tests where the ibex core is stubbed, the ibex tracer still appears to be running. I would like to request that it be enhanced through a runtime plusarg setting (call it "enable_ibex_tracer" or something similar), that is set to 1 by default. Users can override it via plusarg to set it to 0. If set to 0, then the tracer should stay dormant (and it should not even create the log file). 

Thanks
Sri

raulbehl, 2019-10-08:
This should be fixed with #360 
With this implementation in place - the file would only be created if the core retires valid instructions. 

@imphil - Should be okay to mark this as complete? Or you reckon some further updates might be necessary? 
@sriyerg - Hope the above implementation meets your criteria. Basically, gating the clock to ibex would ensure that there is no trace generated hence no trace log as well.

imphil, 2019-10-08:
@raulbehl thanks for reminding me. Yes, this should be fixed. Please re-open if there are additional issues @sriyerg .

closed 2019-10-08 by imphil