lowRISC/ibex #270: DRET treated as normal instruction in M-mode

udinator, 2019-08-29:
[image]
While executing this code in M-mode and with no external debug/irq stimulus applied, the log shows ibex executing a dret and then jumping to `PC=0x00000000`, and the instruction at this address raises an illegal instruction exception, causing a jump to our vectored trap handler at `0x8001c000`.
[image]
In the waveform, we see that ibex recognizes the `dret` instruction, raising `dret_insn_dec`, but then when the core jumps to `FLUSH` state it executes the instruction as normal, and sets the next pc to `PC_DRET`, the value stored in the `dpc` CSR, or `0x00000000`. However according to the debug spec, executing `dret` while not in Debug Mode should cause an illegal instruction exception (Debug Spec, chapter 4.6), which does not seem to be the case in this situation.

davideschiavone, 2019-08-29:
That may be the case for RISCY too, is it @bluewww ?

vogelpi, 2019-08-29:
Thanks for reporting this issue @udinator . I will need to fix the controller to generate the illegal instruction exception when encountering DRET without being in debug mode.

bluewww, 2019-08-29:
@davideschiavone indeed ri5cy also doesnt raise an exception in m-mode.

imphil, 2019-08-29:
@udinator can you confirm that this is fixed now? (And are there tests for it?) (Reopening to get DV confirmation.)

udinator, 2019-08-29:
Confirming that the fix submitted by @vogelpi fixed this issue. DRET instructions are now properly treated as illegal instructions and cause an immediate jump to the vectored trap handler.
@imphil: I do have tests for this in my testbench, and will have a PR out in the near future.

closed 2019-08-29 by udinator

imphil, 2019-08-29:
Thanks a lot!

udinator, 2019-08-29:
No problem!