lowRISC/ibex #285: Dcsr.xdebugver reset value

udinator, 2019-09-05:
Hi all,
I noticed that in the CSR register module, `dcsr.xdebugver` is set to `XDEBUGVER_NO` upon reset, but after reset is over and an explicit write to `dcsr` occurs, the `xdebugver` field will then be permanently set to the value of `XDEBUGVER_STD` for the duration of the simulation.  This seems to contradict with the behavior of the `dcsr.prv` field, which is set to `PRV_LVL_M` at reset, as well as whenever a write occurs outside of reset.  Why does this separate reset behavior for `xdebugver` occur, instead of setting it to `XDEBUGVER_STD` from reset onwards especially since this field is read-only and stays as a constant?
Thank you!
Udi

vogelpi, 2019-09-06:
HI @udinator ,

thanks for creating the issue. I do not see any reason why `dcsr.XDEBUGVER` should not be set to `XDEBUGVER_STD` right at reset. I will create a PR to correct this.
Best Pirmin

closed 2019-09-06 by vogelpi

udinator, 2019-09-06:
Thanks for the fix @vogelpi!