lowRISC/ibex #287: Compile issues while running ibex-tracer example

raulbehl, 2019-09-06:
Hi, 

Ibex top is broken due to the changes in the following commits:

bec84ca2b1
Missing a `,` after the port connection: https://github.com/lowRISC/ibex/blob/60de915d6b85b61a4981263c7a7fa8fa922d3f4d/rtl/ibex_core_tracing.sv#L151

and 

16177fe8db
`NUM_REQS` is not defined: 
https://github.com/lowRISC/ibex/blob/60de915d6b85b61a4981263c7a7fa8fa922d3f4d/rtl/ibex_fetch_fifo.sv#L222

I am surprised to see how CI failed to catch these. The second one probably makes sense since the verilator compilation wouldn't see the assertions due to a guarding `ifndef VERILATOR`. But the first one should have been caught!

raulbehl, 2019-09-06:
Just realised that the azure pipelines don't test the tracing related files. This explains how these were missed by CI. Such issues would be seen once #280 is merged!

raulbehl, 2019-09-09:
As both the issues have been merged to master, It should be safe to close this now? @vogelpi?

vogelpi, 2019-09-09:
Hi @raulbehl ,
thanks for reporting this issue.
Yes, the second part has just been fixed in lowRISC/ibex#292. I think this can be closed now.
Best,
Pirmin

closed 2019-09-09 by vogelpi