lowRISC/ibex #288: Ibex tracing example needs update following the changes in prefetch buffer

raulbehl, 2019-09-06:
The recent changes to I-side logic and prefetch buffer has changed how ibex asserts `instr_req_o`. Due to this the current example in the TB doesn't produce the intended results. The delays in the TB would need to be adjusted as per the new logic.

FYI: Here are the contents of the `trace_core_00000000.log`
```
                Time          Cycles PC       Instr                        Mnemonic
                 240               8 00000080 00008106                          c.mv x2, x1  x2=0x00000000  x1:0x00000000
                 250               9 00000082 00008186                          c.mv x3, x1  x3=0x00000000  x1:0x00000000
                 290              13 00000090 00400113                       addi x2, x0, 4  x2=0x00000004
                 310              15 00000094 ff810113                      addi x2, x2, -8  x2=0xfffffffc  x2:0x00000004
                 340              18 00000098 4000006f                         jal x0, 1024
                 390              23 00000498 039597b3                   mulh x15, x11, x25 x15=0x00000000 x25:0x00000000 x11:0x00000000
                 400              24 0000049c 13410d13                    addi x26, x2, 308 x26=0x00000130  x2:0xfffffffc
                 410              25 000004a0 e1070713                  addi x14, x14, -496 x14=0xfffffe10 x14:0x00000000
                 420              26 000004a4 fff7c793                    xori x15, x15, -1 x15=0xffffffff x15:0x00000000
```
The trace ends abruptly at `420ns` (since now the timing between `instr_req_o`, `instr_gnt_i` & `instr_valid_i` isn't in-sync anymore).

Also, it would be nice if we could come up with a new naming scheme for the dumped filename? The current one has way too many zeros getting appended!

udinator, 2019-09-07:
I am encountering the same issue as well due to the I-side logic changes in #282, my trace log when running `riscv_rand_instr_test` looks like this, abruptly cutting off after around 400 clock cycles:
[image]
Has the documentation been updated to explain how the new timing works?

imphil, 2019-09-07:
>  Has the documentation been updated to explain how the new timing works?

The RVFI interface shouldn't change, this needs to be fixed in the RTL.

PR #293 (2019-09-12): [I-side] - Fix issues found in tracing example
- Fixes #288
- Add missing grant qualification to stop spurious address updates
- Make RTL robust to spurious rvalid signalling
commit f025236a (2019-09-12): [I-side] - Fix issues found in tracing example

- Fixes #288
- Add missing grant qualification to stop incorrect address updates
- Make RTL robust to spurious rvalid signalling
- Make sure a request is held until granted
- Remove incorrect assertion
closed 2019-09-12 by tomroberts-lowrisc