lowRISC/ibex #290: Incorrect irq_fast ID assignment

udinator, 2019-09-07:
Ibex assigns all of the fast interrupts `irq_fast_i[14:0]` an ID from 16 to 30, so that `irq_fast_i[0]` has an ID of 16, `irq_fast_i[1]` has an ID of 17, and so on.  In the [Ibex CSR register file](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv#L572), the entire 15 bit array is output to the [Ibex controller module](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_controller.sv#L165), where `mfip_id` is assigned based on on the value of `irq_fast_i[14:0]`. Later on in the controller, `exc_cause_o` is [assigned](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_controller.sv#L363) by appending 1'b1 to `mfip_id`. This is sent to the [Ibex IF stage](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_if_stage.sv#L109), where the 5 bit `irq_id`, which in this case is the ID assigned earlier in the controller, is multiplied by `0x4` and added to the address held in `mtvec` to get the trap address. However, the ID's of all fast interrupts are still from 0 to 14, which seems incorrect according to the RISC-V Privileged Spec Ch. 3.1.16. This means that there are now conflicts with all other interrupts that have ID's between 0 and 14, so Ibex will now jump to the wrong vectored address for all fast interrupts that are sent to the core. I think the `irq_fast` ID assignments in the CSR file should be from between 16 to 30 to fix this issue and comply with the spec.

udinator, 2019-09-07:
adding @tjaychen for visibility

vogelpi, 2019-09-07:
Hi @udinator ,
thanks for reporting this issue. You are right. The generation of `mfip_id` in the controller is wrong. For example, it should be
   ```if      (csr_mfip_i[14]) mfip_id = {1'b1, 4'd14}; // ID 30```
instead of
   ```if      (csr_mfip_i[14]) mfip_id = 5'd14;```

I will fix it accordingly and create a PR.

closed 2019-09-09 by vogelpi