lowRISC/ibex #296: Instruction address changes before a request/grant handshake

vogelpi, 2019-09-10:
The instruction interface of Ibex allows to change the address while asserting request and waiting for a grant. This behavior is inherited from RI5CY. If I remember correctly, the primary reason for having this in RI5CY is that in multi-core configurations (PULP) with a shared instruction cache, it allows to reduce contention on the cache and leads to notable performance gains. This can also be beneficial for performance in Ibex (e.g. in case the core takes a branch while the instruction interface is still waiting for the grant of a previous request) but has over and over again confused people.

There were several GitHub issues where we needed to explain this to users, but the problem was usually something else. However, we never had an issue to discuss if we want to change this behavior or keep it.

I think we should remove it. Ibex is a different core that does not target multi-core configs with shared I-caches, so the actual performance benefit is probably relatively low. Instead, it is confusing and complicates translation into popular protocols such as AXI/AHB (it is however legal for TileLink interconnects) as mentioned in pulp-platform/riscv#128.

WDYT?

arjanbink, 2019-09-10:
Thank you for adding me to this ticket. I would be strongly in favor of removing this unexpected behavior and make sure that the address is not changed during non-granted requests:

- It unifies the behavior on instruction and data side
- It eases the design of AHB and AXI adapters
- It increases the performance of AHB and AXI adapters
- There is no benefit of using the existing implementation in cases where grants are immediate (which for many scenarios would be the normal case)

I indeed filed the ticket on RI5CY, but I did not have yet taken time to really discuss it with them in more detail. (I did not file it before on Ibex as I noticed it was already discussed in the past)

imphil, 2019-09-10:
Just to add a couple of pointers and anecdotal evidence to that:
- Documentation of the current behavior: https://ibex-core.readthedocs.io/en/latest/instruction_fetch.html#protocol
- Older riscy bug about the same topic: https://github.com/pulp-platform/riscv/issues/50 (that's what caused the documentation note to be added)
- There was more discussion on that with Ashish when doing formal verification, who was equally confused by the current behavior.

PR #332 (2019-10-01): [Prefetch buffer] - various bug fixes
- Fix incorrect address output to mepc on interrupt (fixes #320)
- Fix instruction address changing before grant (fixes #296)
- Remove excess address flops in fetch_fifo
- Remove restriction on outstanding requests
commit f2fccaad (2019-10-01): [Prefetch buffer] - various bug fixes

- Fix incorrect address output to mepc on interrupt (fixes #320)
- Fix instruction address changing before grant (fixes #296)
- Suppress requests and reg writes on fetch error (fixes #340)
- Remove excess address flops in fetch_fifo
- Remove restriction on outstanding requests
closed 2019-10-01 by tomroberts-lowrisc