lowRISC/ibex #301: Core_sleep_o issue

udinator, 2019-09-10:
Hi,
I was running a `wfi` test, expecting the new `core_sleep_o` signal (#284) to be asserted after the `wfi` started being executed, but instead it looks like this signal remains at `1'b0` for the entire test, and switches to `1'bX` at some points  as well, as shown in the waveform below:
[image]
Additionally, there may be an issue with `clock_en`, as it is `1'b1` through the entire simulation, but changes to `1'bX` after a `wfi` is executing.

GregAC, 2019-09-11:
The current implementation of the sleep signal is very simple, it just exposes the inverted clock_en to the outside world.  Hence issues in one will match issues in the other.  I suspect we'll want something more sophisticated eventually (current version may not give good timing for one thing).

As verilator doesn't do X I can't reproduce the issue other than confirm sleep works as I'd expect on an earlgrey verilator simulation (goes high on a WFI).

The issue may be X on other inputs, debug_req_i, irq_nm_i, irq_software_i, irq_timer_i, irq_external_i, irq_fast_i or Xs in the mie_q flop.

tjaychen, 2019-09-11:
@udinator 
let me know if you want me to help take a look at anything.

vogelpi, 2019-09-13:
Hi @udinator ,
thanks for creating the issue. As Greg pointed out we can't see the X in Verilator and cannot really tell what the issue is. If you can forward us the waves generated at your end, we can study them and hopefully find a solution. We would need the waves of the entire processor (`ibex_core.sv`). You can directly upload them here inside a .zip-file.

Best,
Pirmin

udinator, 2019-09-13:
Hi vogelpi and GregAC, I will first update our testbench to ensure there are no X signals being driven/seen anywhere, and if that does not appear to solve the issue, I'll dig deeper into this and try to root cause it before updating this thread. Best, Udi

udinator, 2019-09-19:
Updating this issue to confirm that after modifying the testbench to drive all external stimulus wires to `0` after reset, `core_sleep_o` appears to be working as intended, going high several cycles after a `wfi` starts executing.
Closing this issue for now.

closed 2019-09-19 by udinator