lowRISC/ibex #310: Basic memory transaction latency

vit82, 2019-09-13:
During basic memory transaction Load-Store Unit expects signals data_gnt_i and data_rvalid_i to become active one after another (see Figure 3 at https://ibex-core.readthedocs.io/en/latest/load_store_unit.html). This introduces one additional cycle of access latency.

For better system bus performance it's interesting to allow both data_gnt_i and data_rvalid_i become active simultaneously. I tried such scenario. It doesn't work due to Load-Store Unit FSM architecture.
Is there a chance to fix that?

GregAC, 2019-09-25:
Thanks for the suggestion @vit82.  I suspect practically building a system that gives you immediate data response will be tricky for timing reasons.  Currently the data address output comes straight from the ALU and data coming back from memory goes straight in the register write port.  This gives us two timing paths along the lines of

Reg file read (base address) -> ALU (add) -> data address

data -> Reg file write port

Any system that gives us data response the same cycle we raise the request means we get a path that looks like

Reg file read -> ALU -> data address -> mem read -> data -> reg file write port

This could limit total frequency, sufficiently so that you'd have been better off with the version with the latency but a higher frequency.

Whilst I can't claim this as a general rule it does feel like a long timing path to me that would limit frequency in many cases.

Did you have a specific use case in mind?

vit82, 2019-10-10:
I bear in mind a case when memories are accessed on the clock falling edge while the CPU Load-Store Unit works on the positive edge. 
However due to the fact  Load-Store Unit data/address outputs are coming from combinatorial logic (I've just noticed), it doesn't seem possible to execute memory access within 1 clock cycle. Here some pipe-lining is needed, e.g. Load-Store Unit data/address outputs need to be registered with flip-flops. So the present control of data_gnt_i and data_rvalid_i signals looks optimal.

vogelpi, 2019-10-31:
This question seems to have been answered. I am thus closing this issue now. If you don't think so @vit82 , please comment to re-open the issue again.

closed 2019-10-31 by vogelpi