lowRISC/ibex #322: Compressed tracer load/store issue

udinator, 2019-09-16:
Hi @raulbehl,
I'm seeing some weird behavior with the tracer involving long runs of compressed `lw` or `sw` instructions, below is the disassembled instruction sequence that is getting executed:
[image]
And, below is what Ibex tracer is printing out for the exact same sequence:
[image]

raulbehl, 2019-09-17:
Hi @udinator, 
Thank you for reporting this. The behaviour for compressed load/stores is same as uncompressed load/stores and the issue should be seen for there as well?  I'll push a fix for this soon.

imphil, 2019-09-17:
I'm on it (it's just a swap of the register and the immediate). I'll integrate it into some changes I'm doing on the tracer now.

udinator, 2019-10-04:
Hi @imphil - Regarding the recent updates to the ibex tracer, I'm seeing that for the `c.addi16sp` instruction, the `rd` and `rs1` register values are not being shown, so the log entry for this instruction looks like below:
[image]

PR #373 (2019-10-05): Include rs1 and rd in trace of c.addi16sp
New output looks like this:

118         55      00000154        6121    c.addi16sp      x2,64     x2:0x00000030  x2=0x00000070

Fixes #322
commit d8e4852f (2019-10-05): Include rs1 and rd in trace of c.addi16sp

New output looks like this:

118         55      00000154        6121    c.addi16sp      x2,64     x2:0x00000030  x2=0x00000070

Fixes #322
closed 2019-10-05 by imphil