lowRISC/ibex #341: Local CI tests

MarekPikula, 2019-09-20:
What do you think about making a local CI test solution based on Docker container? Currently that's how I'm testing my changes before commiting. I have simple Dockerfile doing pretty much all the initialisation part from `azure-pipelines.yml` and then I'm executing next commands by hand.

Would anyone be interested in PR with this? I could polish the Dockerfile and write simple script executing all the tests.

asb, 2019-09-20:
Thanks for the suggestion Marek. I think independently of `Dockerfile`s, moving more of test logic that is currently in `azure-pipelines.yml` to a separate script is worthwhile.

For an in-tree Dockerfile we'd ideally have a story on making sure it doesn't become stale. I'm not sure how many people the Dockerfile would help, vs having clear instructions on dependencies and letting people handle those dependencies in the way they prefer (Docker, local installation, VMs, LXD, Podman, ...).

MarekPikula, 2019-09-20:
Well, my solution is basically fully based on Pipelines script, so anyone making changes to Pipelines would also make changes to Dockerfile.

My solution is entirely self-contained, so in order to run tests you just run a Docker image with whatever tool you're using with volume pointing to current directory and everything is done automatically as in Pipelines. We could even build an image for Docker Hub, so it would be just one command to run all tests.

imphil, 2019-09-23:
@MarekPikula Please open a PR if you have something to look at; I'm currently traveling and will go back to look at Ibex CI topics in a week from now after ORConf.

MarekPikula, 2019-09-25:
Sure, I'll polish it today and will post PR.

MarekPikula, 2019-09-25:
Where should I put all the related files? Maybe separate folder `docker` in main tree?

imphil, 2020-11-02:
We do have a docker environment for a while now, closing.

closed 2020-11-02 by imphil