lowRISC/ibex #348: RVFI Interface - load/store memory faults

udinator, 2019-09-24:
Hi all,
Quick question regarding the RVFI, in particular [this section](https://github.com/SymbioticEDA/riscv-formal/blob/master/docs/rvfi.md#memory-access) regarding its set of memory access signals. The description for this whole set of signals specify that they should indicate only valid data, when applicable, but when I monitor these signals in Ibex after injecting a load or store error into the memory response, I see that these buses will still contain the calculated memory address and read/write data and corresponding mask, even though the assertion of the error signal should invalidate the transaction.
I propose a solution where on invalid memory transactions (due to memory faults), the relevant R/W memory address/data signals are driven to 0. This would be similar to Ibex's RVFI implementation of `rd_wdata` and `rd_addr`, which are driven to 0 if `regfile_we` is not high for a given transaction. 
Any thoughts or suggestions would be welcome!
Best,
Udi

GregAC, 2019-09-25:
I think we'd be better off driving to X?  For Verilator this may end up driving to 0 or 1 but I don't think that's an issue.  If the data itself is irrelevant (because the mask doesn't say it's valid) then it doesn't matter what it is.  Using X means we're more likely to uncover times we're accidentally using invalid data.

I'd say the same principle should apply in general (so in particular rd_wdata and rd_addr should X when we're not writing a register) though that goes directly against the RVFI specification linked.  I guess in many cases we are actually writing to a register it's just register 0 (so rd_addr == 0 at least is appropriate rd_wdata should be the value we're chucking away) but there are instructions that truly have no destination register (WFI for example).

tomroberts-lowrisc, 2019-09-25:
From the interface, and therefore from the trace stream, there's no specific indication of a load/store abort. I guess the means of determining it is that the next instruction traced will be the exception handler, and thus the tracer knows to cancel the previous load/store.

To my mind, the most useful thing to do is therefore trace the load/store normally (as we do now). Tracing an address of 0 just means the trace user has no idea what address actually caused the abort. We could drop the rmask/wmask to zero to indicate that no data was actually read/written, but again then the trace user has lost info about the size of transaction attempted.

udinator, 2019-09-25:
I agree with @GregAC that having some method like driving to 'X will make it very clear through the RVFI that an erroneous memory transaction is detected, but also agree with @tomroberts-lowrisc that it might also be useful to leave the address and data untouched, to allow the user to see what memory transaction caused the error. Is there the possibility of adding a signal `rvfi_lsu_err` or something similar that would separately tag a given memory transaction as having raised an error? Something like this will break compliance with the RVFI spec, but I'm not sure I see another way of being able to differentiate between successful and failed memory transactions through the existing RVFI signals.

asb, 2019-09-26:
It might be worth starting a discussion about this on the riscv-formal repo to get feedback from the RVFI spec maintainers.

udinator, 2019-09-27:
@asb - Good idea, I've opened an issue on the riscv-formal repo [here](https://github.com/SymbioticEDA/riscv-formal/issues/32).

udinator, 2019-10-31:
An update: as per the discussion in the riscv-formal repo, the RVFI interface does not have a way to indicate that an instruction/data memory access has faulted, it only can indicate PMP exceptions for the time being. I think in this case, it should be okay to leave Ibex's implementation the way it is, so I will close this issue for now, feel free to reopen this issue if anyone has any more feedback!

closed 2019-10-31 by udinator