lowRISC/ibex #396: [ibex_tracer] Ibex tracer improvements

msfschaffner, 2019-10-11:
This is an issue to collect points that could be improved or "beautified" in the instruction tracer. Please feel free to modify/add to this list.

1) Improve printout alignment of the trace (see example below).
2) Correctly handle invalid instructions (see example below).
3) Add more information about exceptions.

Example snippet where an all-zero instruction is interpreted as `c.addi4spn` instead of marking it as INVALID:
```
23688      11839      1a110354        00000013        addi    x0,x0,0   x0:0x00000000  x0=0x00000000
23696      11843      1a110358        7b302573        csrrs   x10,0x7b3,x0      x0:0x00000000 x10=0x00000032
23700      11845      1a11035c        00100073        ebreak  
23708      11849      1a110360        0000    c.addi4spn      x0,x2,0   x0=0x00000000
23726      11858      1a110808        04c0006f        jal     x0,1a110854       x0=0x00000000
23736      11863      1a110854        7b351073        csrrw   x0,0x7b3,x10     x10:0x00000032  x0=0x00000000
23744      11867      1a110858        00000517        auipc   x10,0x0  x10=0x1a110858
23748      11869      1a11085c        00c55513        srli    x10,x10,0xc      x10:0x1a110858 x10=0x0001a110
``` 

Note: there exist other opensource RISC-V cores that have a tracer (such as [Ariane](https://github.com/pulp-platform/ariane/blob/master/src/util/)). Maybe some synergy could be leveraged here and/or the format could be aligned...

imphil, 2019-10-17:
> Note: there exist other opensource RISC-V cores that have a tracer (such as Ariane). Maybe some synergy could be leveraged here and/or the format could be aligned...

- Neither the RI5CY/old tracer, nor the one in Ariane work in Verilator. This requirement drove the rewrite of this tracer.
- We wanted to use the RVFI signals and don't invent our own tracing signals.
- The format is identical to objdump (even in smallest details).

> This is an issue to collect points that could be improved or "beautified" in the instruction tracer. Please feel free to modify/add to this list.

Side note: Please open one issue per problem/enhancement request. That's the only way to keep issues actionable.

Now to your individual points:
> Improve printout alignment of the trace (see example below).

The trace is tab-separated for easier parsing and alignment with objdump. Depending on your tab width settings, the output is aligned or not.

> Correctly handle invalid instructions (see example below).
> Example snippet where an all-zero instruction is interpreted as c.addi4spn instead of marking it as INVALID:

I've opened #408 with a proposed fix.

> Add more information about exceptions.

Can you open a new issue with more specific examples of what kind of output you're looking for?

closed 2019-10-17 by imphil

towoe, 2019-10-21:
> > Improve printout alignment of the trace (see example below).
> 
> The trace is tab-separated for easier parsing and alignment with objdump. Depending on your tab width settings, the output is aligned or not.
> 
I was also not happy with the output, thanks @imphil for pointing out the width setting. I used `$ tabs 16` to get an aligned output.

PR #408 (2019-10-17): Tracer: Decode an all-zero instruction as INVALID
Fixes #396
commit 1c3903e2 (2019-10-17): Tracer: Decode an all-zero instruction as c.unimp

Fixes #396