lowRISC/ibex #419: ibex_core lint fail.

KwangSon, 2019-10-23:
Hi,

I'm trying to build ibex_core_tracing, got some error.
I tried to find similar issue or goggled, but hard to find solution.

OS : debian 4.9.0-9-amd64
fusesoc : 1.9.3
verilator : 4.020
command : fusesoc --cores-root . run --target=lint lowrisc:ibex:ibex_core_tracing

```
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:ibex:ibex_core_tracing:0.1

verilator -f lowrisc_ibex_ibex_core_tracing_0.1.vc -Wall
%Warning-UNUSED: ../src/lowrisc_ibex_ibex_core_0.1/rtl/ibex_register_file_ff.sv:21: Signal is not used: 'test_en_i'
                                                                                  : ... In instance ibex_core_tracing.u_ibex_core.id_stage_i.registers_i
    input  logic                 test_en_i,
                                 ^~~~~~~~~
                 ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message.
%Warning-UNOPTFLAT: ../src/lowrisc_ibex_ibex_core_0.1/rtl/ibex_controller.sv:101: Signal unoptimizable: Feedback to clock or circular logic: 'ibex_core_tracing.u_ibex_core.id_stage_i.controller_i.ctrl_fsm_cs'
  ctrl_fsm_e ctrl_fsm_cs, ctrl_fsm_ns;
             ^~~~~~~~~~~
%Error: Exiting due to 2 warning(s)
%Error: Command Failed /usr/local/bin/verilator_bin -f lowrisc_ibex_ibex_core_tracing_0.1.vc -Wall
Makefile:16: recipe for target 'Vibex_core_tracing.mk' failed
make: *** [Vibex_core_tracing.mk] Error 10
Traceback (most recent call last):
  File "/home/kwang/.local/bin/fusesoc", line 11, in <module>
    sys.exit(main())
  File "/home/kwang/.local/lib/python2.7/site-packages/fusesoc/main.py", line 583, in main
    args.func(cm, args)
  File "/home/kwang/.local/lib/python2.7/site-packages/fusesoc/main.py", line 268, in run
    flags, args.system_name, args.system, args.backendargs, args.build_root)
  File "/home/kwang/.local/lib/python2.7/site-packages/fusesoc/main.py", line 357, in run_backend
    backend.build()
  File "/home/kwang/.local/lib/python2.7/site-packages/edalize/edatool.py", line 117, in build
    self.build_main()
  File "/home/kwang/.local/lib/python2.7/site-packages/edalize/verilator.py", line 166, in build_main
    self._run_tool('make', args)
  File "/home/kwang/.local/lib/python2.7/site-packages/edalize/edatool.py", line 293, in _run_tool
    except FileNotFoundError:
NameError: global name 'FileNotFoundError' is not defined

```
could you give some tips?
thanks.

imphil, 2019-10-23:
Thanks for your bug report.

You've found a Python 2-only error in edalize, a part of fusesoc. There's nothing that we can do on the Ibex side. The good news is that this error is already fixed in https://github.com/olofk/edalize/commit/dcf5607e6924d28815276148948b6e2a63824d8f, just not part of release yet.

You have two options right now:
- Use Python 3. That's very much recommended. (Use pip2 uninstall fusesoc edalize, and then re-install with pip3.)
- Get the latest version of edalize with `pip2 install -U git+https://github.com/olofk/edalize@master`

KwangSon, 2019-10-24:
Yes it solved. thanks!

closed 2019-10-24 by KwangSon