lowRISC/ibex #423: Add slow multiplier test to CI

MarekPikula, 2019-10-24:
As mentioned in #412, now that we have possibility to choose between slow and fast multiplier in top level it would be nice to have a CI test for this configuration as well.

My question is if the entire test flow should be duplicated (so linter and compliance checks), or just some single step?

imphil, 2019-10-24:
Both options have their merits, but the linter is already a good start.  Maybe let's start with the linter?

You can copy the lint step

https://github.com/lowRISC/ibex/blob/294849bb18e27c0487b7317f53ca878217712ffd/azure-pipelines.yml#L88-L95

and modify the fusesoc call to read
```
       fusesoc --cores-root . run --target=lint lowrisc:ibex:ibex_core_tracing --MultiplierImplementation=slow
```

MarekPikula, 2019-10-24:
It seems that fusesoc cannot use vlogparams via command line, only vlogdefine. I can see `RVFI` and `SYNTHESIS`, but no `MultiplierImplementation` in valid arguments.

imphil, 2019-10-24:
It's just that I forgot that we also need to tell the target to make the parameter available, something along those lines (also for the one without tracing):

```patch
diff --git a/ibex_core_tracing.core b/ibex_core_tracing.core
index 84ca758..f97c328 100644
--- a/ibex_core_tracing.core
+++ b/ibex_core_tracing.core
@@ -66,6 +66,7 @@ targets:
     parameters:
       - RVFI=true
       - SYNTHESIS=true
+      - MultiplierImplementation
     default_tool: verilator
     toplevel: ibex_core_tracing
     tools:
```

MarekPikula, 2019-10-25:
Sorry, I'm not too familiar with either FuseSoC nor Verilator.

Now I have the following error:
```
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:ibex:ibex_core_tracing:0.1

verilator -f lowrisc_ibex_ibex_core_tracing_0.1.vc -Wall
%Error: ../src/lowrisc_ibex_ibex_core_tracing_0.1/rtl/ibex_core_tracing.sv:17: Illegal character in decimal constant: s
%Error: ../src/lowrisc_ibex_ibex_core_tracing_0.1/rtl/ibex_core_tracing.sv:17: Illegal character in decimal constant: l
%Error: ../src/lowrisc_ibex_ibex_core_tracing_0.1/rtl/ibex_core_tracing.sv:17: Illegal character in decimal constant: o
%Error: ../src/lowrisc_ibex_ibex_core_tracing_0.1/rtl/ibex_core_tracing.sv:17: Illegal character in decimal constant: w
%Error: Exiting due to 4 error(s)
%Error: Command Failed /usr/bin/verilator_bin -f lowrisc_ibex_ibex_core_tracing_0.1.vc -Wall
make: *** [Makefile:16: Vibex_core_tracing.mk] Error 10
```

MarekPikula, 2019-10-25:
Seems that Verilator doesn't really like strings. I'm just looking at some code in scr1 project and I've noticed [this](https://github.com/syntacore/scr1/blob/master/src/tb/scr1_memory_tb_axi.sv#L77...L81).

imphil, 2019-10-25:
This looks like a Verilator bug, which typically gets fixed quickly once we report it. If you get around, please report it with a testcase, or otherwise I'll look into it next week.

MarekPikula, 2019-10-25:
Probably the best would be if you reported it since as I said I'm not too familiar with Verilator. Thanks in advance.

imphil, 2020-11-02:
That's resolved now, and we run all supported configurations of Ibex in CI.

closed 2020-11-02 by imphil