lowRISC/ibex #428: RISC-V compliance mtvec alignment

towoe, 2019-10-25:
Follow-up for PR #427 and in connection to #100 

The current RISC-V compliance test places a [trap handler](https://github.com/lowRISC/riscv-compliance/blob/master/riscv-test-suite/rv32i/src/I-EBREAK-01.S#L74) at non 256-byte aligned position and uses this as a target for [mtvec](https://github.com/lowRISC/riscv-compliance/blob/master/riscv-test-suite/rv32i/src/I-EBREAK-01.S#L47).

Ibex requires that the mtvec value is 256-byte aligned.

RISC-V Privileged Spec:

> If mtvec is writable, the set of values the register may hold can vary by implementation. The value in the BASE field must always be aligned on a 4-byte boundary, and the MODE setting may impose additional alignment constraints on the value in the BASE field. 

> An implementation may have different alignment constraints for different modes. In particular,
MODE=Vectored may have stricter alignment constraints than MODE=Direct.

> Allowing coarser alignments in Vectored mode enables vectoring to be implemented without a
hardware adder circuit.

@vogelpi mention in the PR that this should be fixed upstream.
The purpose of this issue is make the problem visible and to discuss the correct solution.

vogelpi, 2019-10-25:
Thanks @towoe for creating this issue :+1:

towoe, 2019-10-25:
> This has also been mentioned as part of another discussion here: [#99 (comment)](https://github.com/lowRISC/ibex/issues/99#issuecomment-519484167)

Ah thanks! The reference/title did not show that this discussion was already happening.
Best place for the issue would then be riscv-compliance to have the best visibility.

rswarbrick, 2021-01-27:
@towoe, @vogelpi: As far as I understand the linked issues and discussion, this is a problem with riscv-compliance requiring something stronger than what is required in the spec. (Is that right?)

I think we're not planning to change our mtvec alignment requirement, so should we just close this (and track any waivers needed as part of #100)?

vogelpi, 2021-01-27:
Hey Rupert,
I lost a bit track of this. Thanks for pinging me.

I think you got it right. However, it seems that riscv-compliance very recently modified the behavior of the corresponding test (see https://github.com/riscv/riscv-compliance/issues/72#issuecomment-747072591) in a way that should allow Ibex to pass if the default `mtvec` doesn't point to ROM. Checking the Ibex riscv-compliance "wrapper" https://github.com/lowRISC/ibex/tree/master/dv/riscv_compliance it seems the default `mtvec` points to RAM, so the test should pass now.

However, I don't know which version of the compliance suite we usually run/is run in CI. I should probably check that.

towoe, 2021-01-27:
There are still problems with riscv-compliance and `mtvec` as mentioned for example in https://github.com/riscv/riscv-compliance/issues/145
But this is an issue with the compliance framework so yes, I agree that this issue can be closed and we use the tracking issue for this.

closed 2021-01-27 by towoe