lowRISC/ibex #429: Parameter of type 'string' is not supported by Cadence Genus

pbing, 2019-10-26:
The new parameter `MultiplierImplementation` is not synthesizable with Cadence Genus.

```
Error   : Unsupported declaration. [VLOGPT-32] [read_hdl]
        : String type in file '../../ibex/rtl/ibex_ex_block.sv' on line 13, column 20.
        : The following declarations are not supported for synthesis:
    - tri0
    - tri1
    - event
    - class
    - chandle
    - SV string type
    - associative array.
    parameter string MultiplierImplementation = "fast"
```

Maybe '0 / 1' instead of '"slow" / "fast"' could be considered instead?

imphil, 2019-10-26:
Can you please give it a try without the "string" type annotation? I.e. just "parameter MultiplierImplementation"?

pbing, 2019-10-26:
This work around could be synthezised.

> 
> Can you please give it a try without the "string" type annotation? I.e. just "parameter MultiplierImplementation"?

— 
Bernd

pcotret, 2019-10-28:
I was playing with @GregAC [example](https://github.com/lowRISC/ibex/tree/master/examples/fpga/artya7-100) for the Arty board.

Got the same problem with Vivado 2018.2 on two files:
- ibex_core.sv
- ibex_ex_block.sv

Synthesis is OK without the **string** type annotation.

imphil, 2019-10-28:
Would you mind opening a PR removing the string annotation?

pcotret, 2019-10-28:
Sure, will do it a bit later.

vogelpi, 2019-10-31:
I am closing this issue now. lowRISC/ibex#430 has been closed but lowRISC/ibex#431 got merged instead. The issue is fixed.

closed 2019-10-31 by vogelpi