lowRISC/ibex #434: how to write interrupt handler

vit82, 2019-10-29:
Question: 
I'm trying to write an interrupt handler. As far as I understand the handler should end with mret command. This command requires ra(x1) register to keep the return address. 
At the same time subroutines in the main program code contain ret command which also uses ra register.
So if I explicitly affect ra in the interrupt handler, the interrupted subroutine will return to the wrong address. If I don't update ra in the interrupt handler explicitly by copying MEPC, the CPU doesn't jump to the address where it was just before the interrupt occurrence.

Probably I miss smth. Please, advice.

GregAC, 2019-10-30:
From the RISCV Privileged spec (Trap Return Instructions section 3.2.2 on the version I have to hand):

'xRET sets the pc to the value stored in the xepc register.'

So the mret doesn't use ra it uses mepc for the address it jumps to.

vit82, 2019-10-30:
Thanks! Just proved your answer by simulation.
Next, I see that IBEX core has a stack of depth 1 to support nested interrupts (e.g. PC is stored in ibex_core.cs_registers_i.mstack_epc_q[31:0]). 

Didn't you consider to provide a Vlog parameter to define the stack depth?

vogelpi, 2019-10-31:
Hi @vit82,
No, we did not consider to add a parameter for this. These stack registers are actually a non-standard extension to enable recoverable NMIs.

The RISC-V specification does not foresee recoverable NMIs. Nested interrupts are still possible without stack registers. As defined in the specification, the core disables interrupts when taking the interrupt, i.e., when entering the handler. This is done in hardware. To handler then needs to save some of the CSRs in software before interrupts can be re-enabled. Obviously, this does not work for Non-Maskable Interrupts (NMIs) as these cannot be disabled, even temporarily. If an NMI arrives while the core is saving these CSRs (before re-enabling interrupts), the current values of the registers would just be overwritten. The processor could not return after handling the NMI and then the original interrupt. That's what the stack registers are for.

Best

vit82, 2019-11-04:
@vogelpi ,
may you share any information of differences in irq_fast_i[14:0] and irq_external_i interrupts?
The user manual contains very little information explaining how CPU treats them. Thanks!

vogelpi, 2019-11-04:
- `irq_external_i` is used to connect a platform-level interrupt controller (PLIC) (such as https://github.com/pulp-platform/rv_plic). The documentation linked at the example PLIC also gives some details on the capabilities of such a PLIC. This usually includes switching on/off individual and a possibly very large number of interrupt sources as well as priority management. To this end, the PLIC has usually a device interface connected to the system bus. That is, the core manages the PLIC like a normal, memory-mapped peripheral.
- `irq_fast_i[14:0]` are just 15 fast interrupts according to the RISC-V spec. These fast interrupts can also be turned on/off individually but there is no priority management. The highest ID has the highest priority. Management of these interrupts happens through the `mie`/`mip` CSRs inside the processor core. These are not mandatory for a RISC-V implementation but Ibex supports them to allow for simple system configurations without PLIC. Please have a look at the privileged spec for more information about these.
Best,
Pirmin

vit82, 2019-11-15:
@vogelpi , may you please, comment the following quotation from RISCV privileged spec:

[image]

Why do the authors claim a two level stack since there's only one instance of xPP/xMPIE present?

vogelpi, 2019-11-15:
Hi @vit82 ,

I interpret this part of the specification as follows:
- For the interrupt enable bit, the two-level stack comprises xIE = the current interrupt enable bit, and xPIE = the previous interrupt enable bit. When an interrupt/exception is taken, xPIE always gets the value of xIE, and xIE is always set to 0 (to allow software to backup CSRs and other registers).
- For the privilege level, there is xPP holding the previous privilege level. The current privilege mode cannot be read from a CSR.

As I understand, there is not a two-level stack to backup these bits/fields, but one level holds the current value and one level the previous value. Whenever nested interrupts are needed, `mstatus` (as well as other CSRs like `epc`) need to be backed up in software as you mentioned. Considering that when taking an interrupt or exception, software anyway needs to backup the 31 general-purpose registers, saving a couple more CSRs probably does not hurt too much. In contrast, additional "stack"/backup registers come at a cost and they mainly give you atomicity, which is needed if you can't disable interrupts (like in the case of NMIs).

Best,
Pirmin

vogelpi, 2019-12-16:
I am closing this issue as there has not been any activity for one month now. Feel free to re-open if you would like to re-discuss or have further questions related to that issue.

closed 2019-12-16 by vogelpi