lowRISC/ibex #450: Interrupt isn't taken in vicinity of a branch instruction

vit82, 2019-11-06:
Below is a screenshot where a fast interrupt handler is mistakenly skipped. 
The CPU jumps to IRQ vector at address 0x1040., there a jump instruction to address is 0x1084 is located. The jump isn't performed, instead main program is executed (addresses 0x00FC -> 0x0100 -> 0x0104 etc )
[image]

Here below is another screenshot where the same interrupt occurs 1 clock cycle later. It can be seen a normal sequence of addresses, i.e. CPU jumps to address 0x1084 where the interrupt handler actually locates.
[image]

I suspect the reason is due to vicinity of branch instruction (red signal branch_set_i).

vit82, 2019-11-06:
The result is produced with RTL commit 275c331

tomroberts-lowrisc, 2019-11-07:
Hi @vit82 , thanks for the detailed report.
Can you tell me the value of ibex_core.instr_valid_id, or better yet, make VCD dump available somewhere for us to have a look at.

The branch_set_i signal shouldn't cause the core to jump to a new address as long as instr_valid_id isn't set (and that should have been cleared by the interrupt).

vit82, 2019-11-08:
Here I've uploaded .trn fie since it's smaller in size vs .vcd (here no more 10MHz is allowed). Hope you can read it. If not, I'll check alternative ways how to upload 35MHz .vcd.
[issue450_bug_seen.trn.tar.gz](https://github.com/lowRISC/ibex/files/3824670/issue450_bug_seen.trn.tar.gz)

The cursor at the screenshot is placed at 10,120us.

imphil, 2019-11-08:
Unfortunately we don't have access to Cadence tools here and are unable to open TRN files. Can you convert it to VCD and upload it either here in compressed form (e.g. .bz2 or .xz), or upload it to dropbox or google drive or something along those lines?

vit82, 2019-11-08:
Alright, here I've uploaded it at my GDrive:
https://drive.google.com/file/d/1iqV3MN7FmzLbDblzmfNRxjjRqJFDuMjy/view?usp=sharing

vit82, 2019-11-08:
[firmware_rv32emc.lss.txt](https://github.com/lowRISC/ibex/files/3825336/firmware_rv32emc.lss.txt)
And here's the firmware listing file.

tomroberts-lowrisc, 2019-11-08:
Thanks, I'll take a look

tomroberts-lowrisc, 2019-11-08:
Hi @vit82 - it looks like you might be hitting some kind of simulator / synchronisation issue.

The controller state (ctrl_fsm_cs) makes an impossible jump to IRQ_TAKEN (on the same cycle as the irq arrives), without a lot of other state changing. The next state signal (ctrl_fsm_ns) for example is never set to IRQ_TAKEN.

How are you driving the irq lines? If they are driven from testbench/software BFMs, are the driving signals clock synchronised?

vit82, 2019-11-11:
Ok, it doesn't work when I assert the interrupt line upon clk posedge. Like this:
	initial begin
		repeat (12) @(posedge dut_top.clk);			                   //#120ns;
		force tb.dut_top.riscv_dut.dut.rv32emc.uut.ibex_core.irq_fast_i[0] = 1'b1;
		repeat (3) @(negedge dut_top.clk);
		force tb.dut_top.riscv_dut.dut.rv32emc.uut.ibex_core.irq_fast_i[0] = 1'b0;
      end

Though it works if I do the same on the clk negedge with pulses 3 clock period wide or more.
With pulses 1 or 2 clock periods, it doesn't work even on clk negedge.

tomroberts-lowrisc, 2019-11-11:
Hi @vit82, all interrupts in Ibex are level sensitive. Practically, this means that you must hold the interrupt line asserted until the interrupt is taken and handled by the CPU. Normally, the interrupt handler routine would then write to a register to acknowledge the interrupt, at which point the line can be de-asserted.

vogelpi, 2019-11-11:
Just for reference, this was also discussed in lowRISC/ibex#199.

vit82, 2019-11-12:
Thank you. 
The interrupts generated on clk negedge are taken correctly. Also no problem occurs if interrupts are produced upon clk posedge with a unit delay added:
@(posedge clk);
#1;

Still there's a problem if an interrupt is set immediately upon clk posedge.
Now I realize that the root cause is delta delay added to the CPU clock by ibex_core.core_clock_gate_i. 
So the user should account for this fact when creating stimuli.

closed 2019-11-13 by vit82