lowRISC/ibex #467: Optional U-Mode

MarekPikula, 2019-11-13:
Would it be feasible to make U-Mode enable a core parameter like PMP?

I'm sure there are some use cases where it's not really needed, yet as small logic utilization as possible is required. Is the U-Mode so tightly integrated into the core that it cannot be easily separated as a parameter?

tomroberts-lowrisc, 2019-11-13:
We didn't parameterize it as it has such a small overhead (just a handful of extra flops). Is there a particular part where you think area could be saved?

MarekPikula, 2019-11-13:
Not really. To be honest I thought the overhead is bigger than a handful of extra flops.

tomroberts-lowrisc, 2019-11-13:
Yeah it's surprisingly small. If you're interested, all of the changes to implement it are in this commit 2aacd2b98be61fa4c9b633d1ddba9d1d5c894faf

MarekPikula, 2019-11-13:
Looks really good :+1: Yeah, so in the end it's indeed not needed to make it a parameter. Thanks for quick response.

closed 2019-11-13 by MarekPikula