lowRISC/ibex #469: Failure with 4 nested interrupts

vit82, 2019-11-13:
I'm trying to play with nested interrupts. So I made a basic simulation where 4 fast interrupts irq_fast_i[3:0] are generated in a row. 
Upon exit from the 4th (last) interrupt routine, the simulation crashes since PC gets mtvec value.
Besides I noticed that MPP unexpectedly (in my opinion) gets PRIV_LVL_U taken from mstack_q.mpp upon exit from 2nd interrupt.

[image]

vit82, 2019-11-13:
Here's link to VCD of the whole simulation:
https://drive.google.com/file/d/1_tCNmXmqjzP6HRwKA9owZvmAPoSzf3ov/view?usp=sharing

tomroberts-lowrisc, 2019-11-13:
Thanks for the detailed report @vit82 - You're right it's incorrect for the core to jump to U-mode. Looking at it, I'm not sure how nested interrupts are possible with these non-standard mstack registers since they can't be written to.
I'll look into it.

sjgitty, 2019-11-13:
@udinator can you comment on the testing to date of nested interrupts?

udinator, 2019-11-13:
@sjgitty - I haven't tested nested traps at all as of right now, that is something that I will now be testing along with U-mode and PMP.

tomroberts-lowrisc, 2019-11-14:
Hi @vit82 I looked into the waves and your sample code.
I believe the problem will be fixed by also saving the `mstatus` register to the stack (along with `mepc`) before re-enabling interrupts. The `mstatus` register should then be restored before executing `mret`.

Ignoring the non-standard `mstack` registers for a moment, the RISCV priv spec states that on executing an `mret` instruction, `mstatus.MPP` should be set to U-mode. So in nested interrupts, executing multiple `mret` instructions without restoring the previous interrupt context will drop you down into U-mode.

The `mstack` registers effectively do this for you for one level of nesting, but for further levels the software must handle it by stacking the interrupt context.

Apologies for the delay in getting back to you, there doesn't seem to be any decent example available for how to handle nested interrupts!

vit82, 2019-11-14:
Thanks! Putting the register mstatus on stack solved the issue!

closed 2019-11-14 by vit82

imphil, 2019-11-14:
@vit82 great that it works! Would you mind sharing a bit of example code that shows the entry/exit from a interrupt when using nested interrupts? That's something we really should put into our user guide. Or even better, would you mind doing a pull request with some documentation, e.g. added to https://github.com/lowRISC/ibex/blob/master/doc/exception_interrupts.rst?

vit82, 2019-11-15:
> @sjgitty - I haven't tested nested traps at all as of right now, that is something that I had planned to work on after U-mode verification is done.

May you, please, share the status of IBEX verification activities? Is there an access to verification plan?

udinator, 2019-11-15:
Hi @vit82 - As our verification of Ibex core is part of the larger OpenTitan verification effort, a DV status for the core can be found [here](https://github.com/lowRISC/opentitan/blob/master/hw/ip/rv_core_ibex/doc/checklist.md), and complies with the OpenTitan Verification milestone checklist, found [here](https://github.com/lowRISC/opentitan/blob/master/doc/project/checklist.md).
Regarding the verification plan, as we are verifying the core by using the Google developed [RISC-V Instruction Generator](https://github.com/google/riscv-dv) as explained in the verification documentation, the existing regression list is found [here](https://github.com/lowRISC/ibex/blob/master/dv/uvm/riscv_dv_extension/testlist.yaml).
Feel free to let me know if you have any question/concerns!

vit82, 2019-11-19:
Thanks, @udinator , is there a possibility to track progress of verification? In the regression list you pointed I couldn't find status of the test cases (pass/fail) or other metrics (e.g. functional coverage).

By the way, is there a viewer to display conveniently yaml files?

udinator, 2020-01-16:
Hi @vit82 , apologies for late response. As of right now, there is no progress/coverage tracker for Ibex verification, the regression list just displays the test case configuration to pass into the simulation flow.
As far as yaml files, I just open them in a text editor to display them, as it is just another formatting language similar to hjson format.