lowRISC/ibex #473: MPHP Removal Mechanism

Sarafim, 2019-11-14:
It would be nice to add some ability mechanism to remove mphp registers(mhpmcounter_q) from the design via some parameter or defines. I just have synthesised the core with and without them, and can established that mhpm register occupy about 35% of the total area, but they can be optional for some tasks.

imphil, 2019-11-14:
Have you synthesized with `MHPMCounterNum = 0`? (https://github.com/lowRISC/ibex/blob/master/rtl/ibex_core.sv#L17)

Sarafim, 2019-11-15:
yes, but it removes only flip-flops in range of
ibex_core/cs_registers_i/mhpmcounter_q_reg[\*][0] : ibex_core/cs_registers_i/mhpmcounter_q_reg[\*][31]
and doesn't influence 
ibex_core/cs_registers_i/mhpmcounter_q_reg[\*][32]-ibex_core/cs_registers_i/mhpmcounter_q_reg[\*][63]

tomroberts-lowrisc, 2019-11-15:
Looks like this could be coded differently to guarantee they get synthesised out

imphil, 2019-11-15:
@Sarafim Do you have code that optimizes the flops out fully? Would you mind opening a pull request with that? Please let me know if you get stuck anywhere in the process, we're happy to help you getting started.

Sarafim, 2019-11-15:
Unfortunately, I do not have any solution for this problem.
To see the area without the MPHP registers I just commented them.

vogelpi, 2019-11-18:
I just checked Vivado, it optimizes away the unused counters/flops correctly. @Sarafim, could you maybe give some information on the synthesis tool you are using?

Sarafim, 2019-11-18:
it is Genus from Cadence

Sarafim, 2019-11-19:
I have discussed this issue with Cadence support. To optimize all MPHP registers correctly the syn effort should be set as "high". It is true at least for genus 17.12.
Thanks!

vogelpi, 2019-11-19:
Thanks @Sarafim , that's good to know!

MarekPikula, 2019-11-20:
I guess it would be the best to include this information somewhere in documentation.

closed 2019-12-11 by tomroberts-lowrisc

GregAC, 2020-01-07:
@Sarafim please note I've just made a fix to Ibex that should mean the unused performance counters won't be included regardless of optimisation level see commit https://github.com/lowRISC/ibex/commit/9c6ddae626ac22eb734be81509b9b7456174d8a5

PR #516 (2019-12-11): [doc] Add a comment on mhpmcounter optimization
- fixes #473
commit b2bbe507 (2019-12-11): [doc] Add a comment on mhpmcounter optimization

- fixes #473
PR #541 (2020-01-07): [rtl] Don't elaborate unused mhpmcounter flops
Refactors performance counters so only flops that are required from the
given parameters are explicitly elaborated without relying on
optimization to remove unused flops.

Fixes #473

The previous RTL produced a circuit where the value of all mhpmcounter flops would be maintained, unused bits would be reset to 0 and written 0 but before reset they could be some other value. Whilst it was possible for synthesis tools to decide this meant they could be ignored and seen as constant 0 values it requires setting optimization to 'high' or similar in som [... cut]
commit 9c6ddae6 (2020-01-07): [rtl] Don't elaborate unused mhpmcounter flops

Refactors performance counters so only flops that are required from the
given parameters are explicitly elaborated without relying on
optimization to remove unused flops.

Fixes #473