lowRISC/ibex #482: An higher prio interrupt routine is interrupted by lower prio IRQ

vit82, 2019-11-19:
When executing a higher priority interrupt routine, it can be interrupted by a lower prio interrupt if MIE is set high and high prio IRQ source cleared. Is it an intended implementation? 

Below is a screenshot where fast_irq_i[0] and fast_irq_i[1] occur simultaneously. First, fast_irq_i[1] is being served. But after I clear fast_irq_i[1] source & set MIE high, the routing is preempted by fast_irq_i[0] . Once fast_irq_i[0] routine is complete, the rest of fast_irq_i[1] is completed.
[image]

vit82, 2019-11-19:
[firmware_rv32imc.lss.txt](https://github.com/lowRISC/ibex/files/3864062/firmware_rv32imc.lss.txt)
Attaches is FW listing file.

vogelpi, 2019-11-19:
Hi @vit82 ,
yes, that is intended behavior. As far as I know, the priority only matters if multiple interrupt lines are high and the core needs to decide which one to start handling first.

vit82, 2019-11-19:
Alright, I see it's possible to implement the above mentioned function by adding extra logic outside CPU. But it would require replication of arbitration logic etc. So isn't it more optimal to put everything inside CPU core? Or may there's a use case when higher prio routine needs to be preempted by a lower prio routine?

vogelpi, 2019-11-19:
This behavior can be implemented in software: Before re-enabling interrupts by setting `mstatus.MIE` to 1'b1, the content of `mie` has to be modified. This CSR holds the interrupt enable of the individual interrupts.

For example when handling fast interrupt no. 1 (`mfip[1]`, interrupt ID 17), all `mie[17:0]` has to be set to 0. The interrupt routine is then only interrupted by fast interrupt no. 2 - 14 (ID 18 - 30).

vit82, 2019-11-21:
Ok, your suggested approach works. Though it increases interrupt service overhead.

closed 2019-11-21 by vit82

imphil, 2019-11-21:
Please look into an external interrupt controller, like PLIC or CLIC, if you are interested in more hardware processing for your interrupts.