lowRISC/ibex #485: Regression testing failed

NCGNT, 2019-11-21:
I've tried to run regression test but it has failed a few times. I debugged one of the fails and found differnce between iss_sim and rtl_sim program flows.
So, at asm program there is a jump to subroutine. Trace log of this jump looks the same for both iss and rtl simulations but return code differs.

Subrotine call code at main program looks like this:
```
                  ...
                  j init_user_mode
```
And soubroutine code is:
```
init_user_mode:   
                  ...
                  mret
```
According to trace of ISS simulator it uses "mret (0x30200073)" command as the return instruction, but RTL uses "jal(0800006f)" in the same case. So, they used different return codes which causes jumping to different addresses.
As I've seen ISS uses .o file as the source and rtl simulator .bin which can be a root of the problem, but both of them are generated by gen rule of makefile from the same source.

udinator, 2019-11-21:
Hi @TAD-dev - Thanks for reporting this! It would appear that in the `init_user_mode` section, there is an access to `sstatus`, which Ibex does not support. Looks like this is an issue with the instruction generator, rather than the core. I'll prepare a fix in the generator and update the vendor directory accordingly.

udinator, 2019-11-21:
@TAD-dev  - #486 and #487, once merged, should fix this issue; it was mostly fixes on the flow side. Thanks again for raising this issue!

udinator, 2019-11-22:
Can you sync to the master branch and try running this test again? Everything should be fixed now.

TAD-dev, 2019-11-22:
After update I've got an error related to randomization on gen step. Please find error message of this error attached below:
```
if (MSCRATCH inside {implemented_csr}) {
                                             |
ncsim: *E,RNDCNSTE (ibex/vendor/google_riscv-dv/src/riscv_rand_instr.sv,108|45): Randomization constraint has this error, which will cause the randomize function to return 0 and no new rand values will be set:                                            
Unsupported parameter type in constraint.
        `DV_CHECK_RANDOMIZE_WITH_FATAL(branch_instr, category == BRANCH;)
                                                                        |
ncsim: *W,SVRNDF (ibex/vendor/google_riscv-dv/src/riscv_directed_instr_lib.sv,323|72): The randomize method call failed. The unique id of the failed randomize call is 808.                                                                                  
Observed simulation time : 0 FS + 44
UVM_FATAL ibex/vendor/google_riscv-dv/src/riscv_directed_instr_lib.sv(323) @ 0: reporter [instr_stack_enter] Check failed (branch_instr.randomize()) Randomization failed! 
```

udinator, 2019-11-22:
Could you please re-sync to master and re-run the test?

TAD-dev, 2019-11-25:
I still have the same error.

udinator, 2019-12-03:
Are you using IUS or Xcelium  to run the simulations?

TAD-dev, 2019-12-04:
This error happens on "gen" step where ius is used for [compilation](https://github.com/lowRISC/ibex/blob/master/vendor/google_riscv-dv/yaml/simulator.yaml#L36) and [simulation](https://github.com/lowRISC/ibex/blob/master/vendor/google_riscv-dv/yaml/simulator.yaml#L45).

closed 2019-12-05 by TAD-dev