lowRISC/ibex #489: Can we run Random test generator with Model sim?

madhava-sahu, 2019-11-22:
Hi all,

I installed the ibex SV/UVM instruction generator for RISCV processor verification. 
I have done this:
Here I have Models sim. Can it be compatible to implement this.

udinator, 2019-11-22:
Hi @madhava-sahu, Questa is supported for the RISCV-DV instruction generation. 
See https://github.com/google/riscv-dv/issues/262 for a similar issue with the Questa setup for RISCV-DV. Have you set the `QUESTA_HOME` environment variable?

madhava-sahu, 2019-11-23:
@udinator Thanks for the reply.

danghai, 2019-11-26:
@madhava-sahu You use the `modelsim_ase` not `questa`. Could you check `verilog_src/uvm-1.2` in your directory?
You should remove `modelsim.ini` when you rerun. You also use `verror <number-error>` to see more detail about error. Ex: in this situation, `verror 25`.

madhava-sahu, 2019-11-26:
Hi @danghai,
Thanks for the reply. 
Yes, uvm-1.2 is in the modesim_ase, verilog_src folder. I executed the run.py after deleting the modelsim.ini. 
I am getting errors like uvm headers not found.
[image]

Please help me..

danghai, 2019-11-26:
Could you provide the `compile.log` in your output directory?

madhava-sahu, 2019-11-26:
Hi @danghai 

command: Python3 run.py --test riscv_arithmetic_basic_test --simulator questa
Please find the compile.log
[compile.log](https://github.com/lowRISC/ibex/files/3891215/compile.log)

danghai, 2019-11-26:
It cannot find  find `include file "uvm_macros.svh" in directories:
Could you try run vmap command again. Ex: 
```
vmap mtiUvm /home/danghai/questasim/uvm-1.2
```

madhava-sahu, 2019-11-27:
Hi @danghai 
Thanks for the reply...
We are using Modelsim.
In this we are also using the same command from the simulator.yaml.
vmap mtiUvm $QUESTA_HOME/verilog_src/uvm-1.2
export QUESTA_HOME=/home/sevyamm/intelFPGA/16.1/modelsim_ase (bashrc)
The output is coming as mentioned in figure.
[image]

Please suggest

danghai, 2019-11-27:
It should
```
export QUESTA_HOME=/home/sevyamm/intelFPGA/16.1/modelsim_ase
vmap mtiUvm $QUESTA_HOME/uvm-1.2 
```

madhava-sahu, 2019-11-27:
Hi @danghai 

As i told you that I have modelsim.
In the installed directory of the modelsim has this path:
/home/sevyamm/intelFPGA/16.1/modelsim_ase/verilog_src/uvm-1.2

Can't we move further with the modelsim?

danghai, 2019-12-04:
@madhava-sahu Sorry for late response. I am not sure about `modelsim_ase`. I usually use `questa`.
Mentor released the new tool named `qrun` (version 2019.4). It provides pre-built flows for coverage, debug, UVM and other flows. You should try to use it when you have the chance :)

vogelpi, 2020-03-03:
There hasn't been any activity in this issue for 3 months now. I am thus closing it. Feel free to re-open it if necessary.

closed 2020-03-03 by vogelpi