lowRISC/ibex #492: MEPC written on `mret` instructions

udinator, 2019-11-22:
Hi all,
I noticed that on an `mret` instruction, `mepc` will get overwritten with the contents of `mstack_epc` as shown below:
[image]
This is causing some test failures, as this update is not reflected in the logs, resulting in a register write mismatch since the Ibex `mepc` now holds `0`, while the `mepc` in the ISS will still contain the correct value that was explicitly written at an earlier point in the simulation (`0x80000144`).
`mepc` will still get written correctly whenever the core takes a trap, it is just this extra write that occurs on `mret` that modifies its state. According to the spec, `mepc` should never be written by the RTL apart from taking traps, so how does this align with the nonstandard `mstack` CSRs that Ibex implements?
Best,
Udi

GregAC, 2019-11-25:
I think the issue is the csr_restore_mret_i signal. It's asserted on any mret but should only be asserted when we're in nmi mode (as per https://github.com/riscv/riscv-isa-manual/issues/261):

https://github.com/lowRISC/ibex/blob/bb7acbdb7a1c1240cf93787396ea1f4812ca3d5e/rtl/ibex_cs_registers.sv#L547-L557

https://github.com/lowRISC/ibex/blob/bb7acbdb7a1c1240cf93787396ea1f4812ca3d5e/rtl/ibex_controller.sv#L549-L556

@vogelpi as you implemented this initially do you agree? I'd be happy to do the PR to fix.

vogelpi, 2019-11-25:
Thanks @udinator for opening the issue and @GregAC for your suggestion. This makes sense.

But I think `csr_restore_mret_i` should not be changed as `priv_lvl_d`, `mstatus` need to be changed independently of `nmi_mode_q`. We should probably feed the `nmi_mode_q` to `ibex_cs_registers.sv` as we do for `debug_mode_q` and then only use this to decide what should happen with `mepc_d`. WDYT?

GregAC, 2019-12-11:
You're right can't just alter the behavior of the `csr_restore_mret_id_o` signal. I'll do a PR with what you suggest, feeding `nmi_mode_q` into `ibex_cs_registers` and ensure the mret behaviour is altered appropriately.

PR #518 (2019-12-16): [RTL] Only restore from mstack in nmi mode
Fixes #492

@udinator could you run the tests you saw failing in #492 again on this?
commit 328aabb5 (2019-12-16): [RTL] Only restore from mstack in nmi mode

Fixes #492
closed 2019-12-16 by GregAC