lowRISC/ibex #495: New bit manipulate instruction request

esshiu, 2019-11-26:
We're interested in the following instructions. They're not currently in RISC-V ISA extension.

**fls (find the largest bit set)**. eg 31 - clz
bits (bit-set-by-position) and bitsi (immediate version)
bitc (bit-clear-by-position) and bitci (immediate version)

Use cases:
fls(Find largest bit set; fls rd, rs). This and bitc are used in the inner loop to find-and-extract bits.
Equivalent to: 
clz rd, rs
xori rd, rd, 31

**bits (bit-set-by-position) and bitsi (immediate version)**
bits rd, rs1, rs2
bitsi rd, rs, IMM

Set bit by index.  rd = rs1 | (1 << rs2)
Equivalent to:
li rs1, 1
sll rs1, rs1, rs2
or rd, rs1, rs2

The encoding for bitc and bits should be identical other than the value to place in bit position rs2.  This may reduce gate count.

**bitc (bit-clear-by-position) and bitci (immediate version)**
bitc rd, rs1, rs2
bitci rd, rs, IMM

Clear bit by index.  rd = rs1 & ~(1 << rs2)
General case:
li rs1, 1
sll rs1, rs1, rs2
xori rs1, rs1, -1
and rd, rs1, rs2

If the immediately preceding instruction was fls rs2, rd, then we know the bit is set in the destination already:
li rs1, 1
sll rs1, rs1, rs2
xori rd, rs1, -1

Allowable in bitwise spec, which indicates it may be useful for lower-performance cores like ours which can't fuse macro ops

GregAC, 2019-11-27:
From the working draft here: https://github.com/riscv/riscv-bitmanip/blob/master/bitmanip-0.92.pdf

sbset, sbclr and the their immediate versions implement the instructions you've given the bits/bitc mnemonics.

I don't immediately see a FLS equivalent but from your desired use-case 'the inner loop to find-and-extract bits' does the order matter (i.e. from high bit to low bit or vice-versa) as you do can do such a loop going from low bit to high bit with the CTZ (count trailing zeros) instructions with the same number of instructions as FLS going from high to low would use.

If high to low order is essential how much does the extra sub instruction really impact overall performance? Take a tight loop of only going through all of the set bits and clearing them one by one from high to low:

```Assembly
li t2, 32

loop:
clz t1, t0
beq t1, 32, end 
sub t1, 31, t1
bitc t0, t0, t1
j loop

end:
```

The extra sub increases execution time by 25%. However presumably you'll actually be doing something with that bit position number which will add instructions to the loop and this won't be the only thing the program is doing so that extra instruction may quickly become insignificant.

I'd be reticent to start adding custom instructions unless we've got some decent evidence of what performance benefit they bring.

Whilst the bit manip spec isn't finalised I don't think it's going to change radically so it shouldn't be too hard to match whatever we've got with the final spec when it's done. It has some suggestions as to how it will be partitioned up. Implementing the Zbb and Zbs (base and single-bit) parts should cover what has been specified in this issue (I suspect most of the change will be happening in the more complex matrix and permutation stuff too which aren't in Zbb and Zbs).

esshiu, 2019-11-27:
Randall,

It seems only FLS is missing from the latest bit manipulation instruction.
Can you please review Greg’s comment below?

Thanks.

Eric

esshiu, 2019-11-27:
From the RTOS side we could use CTZ and loop over bits from low to high in
many places (enabled timers, etc.).  It would be inconvenient to have to
reorder task priorities so 0=high instead of 0=low, since we make some
assumptions that task_id=0 is the idle task.

Note that "sub t1, 31, t1" is 2 instructions, not one:

sub t1, zero, t1
addi t1, t1, 31

or requires an additional register:
(outside loop)

li t2, 31

(inside loop)

sub t1, t2, t1

since there isn't a subtract-from-immediate opcode.

It's probably not the end of the world, given that you have clz/sbset/sbclr.

- Randall.

esshiu, 2019-12-02:
It's okay to close the issue per Randall's comment above.

-Eric

closed 2019-12-02 by esshiu