lowRISC/ibex #496: Verilator ELF loading: zero BSS segment

imphil, 2019-11-27:
Currently, the ELF loader in verilator_sim_ctrl doesn't zero the BSS segment:

https://github.com/lowRISC/ibex/blob/ce7e38351e28bf54cbc7c2856ce6b65b9d2a3fbb/dv/verilator/simutil_verilator/cpp/verilator_sim_ctrl.cc#L721

More discussion here: https://github.com/lowRISC/ibex/pull/411#issuecomment-555924017

towoe, 2019-12-04:
The `crt0.S`s in `examples/sw` clear the BSS section.
This is based on the linker script to provide the start and end position of BSS and of including the zero-ing to the initializing sequence.
So I assume we can require that the loaded software is responsible to set the zeros itself.
Would a readme for simutil_verilator make sense to you @imphil?
This would for now just contain the requirement and the TODO would be deleted.

imphil, 2019-12-04:
Yes, sounds like a plan.

closed 2019-12-05 by imphil