lowRISC/ibex #497: Cadence simulator plusargs command line argument parsing error

TAD-dev, 2019-11-27:
Verification environment uses SystemVerilog line argument for getting signature address [(core_ibex_env_cfg.sv)](https://github.com/lowRISC/ibex/blob/ce7e38351e28bf54cbc7c2856ce6b65b9d2a3fbb/dv/uvm/env/core_ibex_env_cfg.sv#L30). Value is passed to environment at [(makefile)](https://github.com/lowRISC/ibex/blob/ce7e38351e28bf54cbc7c2856ce6b65b9d2a3fbb/dv/uvm/Makefile#L132) at run_sim step. The problem is Cadence simulator cannot recognize this argument as hexadecimal number:
```
void'($value$plusargs("signature_addr=%0h", signature_addr));
                    |
ncsim: *E,SYSFMT (ibex/dv/uvm/env/core_ibex_env_cfg.sv,30|24): $value$plusargs --  value found '0x8ffffffc' for plusarg 'signature_addr=' does not match specified format.
```
This issue can be fixed with something like atohex() SystemVerilog function (this function should be supported by all simulators).

imphil, 2019-11-27:
Thanks for this bug report. Would you mind doing a pull request with your suggested change?

TAD-dev, 2019-12-02:
Yes, I'm gonna to create one. Sorry for late response.

closed 2019-12-04 by udinator