lowRISC/ibex #503: atal: ** Error: (vsim-3828) Could not link 'vsim_auto_compile.so': cmd = '/home/sevyamm/intelFPGA/16.1/modelsim_ase/gcc-4.7.4-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/sevyamm@sevyamm-Lenovo-ideapad-320-15AST_dpi_14144/linuxpe_gcc-4.7.4/vsim_auto_compile.so"  "/tmp/sevyamm@sevyamm-Lenovo-ideapad-320-15AST_dpi_14144/linuxpe_gcc-4.7.4/exportwrapper.o"

eshita18, 2019-11-30:
I am trying to simulate testcase riscv_arithmetic_basic_test in uvm env in the path /opentitan-master/hw/vendor/lowrisc_ibex/dv/uvm in modelsim 10.5b but I am getting this fatal error : Fatal: ** Error: (vsim-3828) Could not link 'vsim_auto_compile.so': cmd = '/home/sevyamm/intelFPGA/16.1/modelsim_ase/gcc-4.7.4-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/sevyamm@sevyamm-Lenovo-ideapad-320-15AST_dpi_14144/linuxpe_gcc-4.7.4/vsim_auto_compile.so"  "/tmp/sevyamm@sevyamm-Lenovo-ideapad-320-15AST_dpi_14144/linuxpe_gcc-4.7.4/exportwrapper.o" 
Can somebody suggest me how to remove this??

These are the commands I am using for simulating it : 
- vmap mtiUvm /home/sevyamm/intelFPGA/16.1/modelsim_ase/verilog_src/uvm-1.2
- vlog -64 +incdir+/home/sevyamm/intelFPGA/16.1/modelsim_ase/verilog_src/uvm-1.2/src +incdir+/home/sevyamm/Desktop/opentitan-master/hw/vendor/lowrisc_ibex/dv/uvm/riscv_dv_extension +incdir+/home/sevyamm/Desktop/opentitan-master/hw/vendor/lowrisc_ibex/dv/uvm/common -printinfilenames -access=rwc -f /home/sevyamm/Desktop/opentitan-master/hw/vendor/lowrisc_ibex/dv/uvm/ibex_dv.f -sv -mfcu +define+UVM_NO_DPI /home/sevyamm/Desktop/opentitan-master/hw/vendor/lowrisc_ibex/dv/uvm/tb/core_ibex_tb_top.sv -l /home/sevyamm/Desktop/opentitan-master/hw/vendor/lowrisc_ibex/dv/uvm/out/instr_gen/compile.log
- vsim -64 core_ibex_tb_top +UVM_TESTNAME=riscv_arithmetic_basic_test -sv_seed 1342

imphil, 2019-12-02:
@eshita18 I see you're using the same home directory as in #489. How are the two issues related?

eshita18, 2019-12-03:
We are trying to run lowRISC ibex env in Modelsim. #489 is related to run random test generator which we were trying to do in modelsim only.
Can you tell me something about the fatal I am getting now??

raulbehl, 2019-12-03:
As far as I am aware the mentioned modelsim version (modelsim 10.5b) doesn't support UVM. What I know is that Modelsim supports all SystemVerilog/Verilog features except:
- SystemVerilog assertions
- SystemVerilog coverage
- randomize method
- program blocks

The above language features are available in Questasim. 
One could still compile UVM based testbenches but without the randomize method support most of the things wouldn't work.

eshita18, 2019-12-03:
Yeah, UVM is supported in modelsim except of these features.
For now, we are trying to simulate a simple adder env with DPI in questa and modelsim. In both, it is giving same fatal. Is it because of DPI??
This is my code:

#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>
#include <errno.h>
//#include <netinet/in.h>
//#include <arpa/inet.h>
//#include <fcntl.h>
#include "dpi_header.h"
#include "svdpi.h"
 int value(int a)
{
   int temp_value = a + 5;
   printf("sum of value coming from SV Env with 5 is ......  %d", temp_value);

return temp_value;

}

----------------------------------------------------------------------------------
This is the top.sv file :
`include "uvm_pkg.sv"
import uvm_pkg::*;
**import "DPI-C" context function int value(int a);**

vogelpi, 2020-03-03:
There hasn't been any activity in this issue for 3 months now. I am thus closing it. Feel free to re-open it if necessary.

closed 2020-03-03 by vogelpi

hirooih, 2020-08-06:
https://github.com/pulp-platform/pulpino/issues/110#issuecomment-326100024
Add '-dpicpppath /usr/bin/gcc' to vsim option.

I hope this helps.