lowRISC/ibex #514: Getting Timeout Error with spike while running random test generator..

madhava-sahu, 2019-12-06:
Hi all,

I am getting timeout errors with spike while executing the random test generator.
command: python3 run.py --test riscv_csr_test --simulator questa
[image]
[image]
Please help me..

udinator, 2019-12-06:
Hi @madhava-sahu - thanks for reporting this! I will look into it.

udinator, 2019-12-06:
I ran the command `python3 run.py --test riscv_csr_test --simulator questa` that you pasted above, and it ran fine, without spike simulator timing out. This test is configured to not run ISS simulation at all, so Spike should not even run at all, so I'm not sure why it's timing out, can you paste what the riscv_csr_test entry in the testlist looks like? There should be a field `no_iss: 1` that disables ISS from running on the compiled test file, which `run.py` checks for inside of `iss_sim(...)`

madhava-sahu, 2019-12-07:
Hi  @udinator 
Thanks for the reply. 
I have checked as per your inputs. I was changed the no_iss: 0. 
In the base_test_list.yaml  file, there are available settings are mentioned below
[image]
I executed the same command: python3 run.py --test riscv_csr_test --simulator questa

Again I am getting same timeout error..
[image]
Please help me.

udinator, 2019-12-07:
Can you change no_iss to 1 and rerun the test? This test is not going to work correctly if no_iss is set to 0.

madhava-sahu, 2019-12-07:
Hi @udinator 

I changed no_iss: 1 and reran it.
The output is:
[image]
In this case, iss is found but it not started to execute the binaries. 
what should i do?

udinator, 2019-12-07:
That is perfectly fine because this test does not need ISS to run, as it is completely self checking and provides it's own checking mechanism unlike the other tests that utilize the full ISS flow. I can modify the run script too avoid printing out the messages relating to ISS for this csr test to prevent confusion. No ISS log should be produced for this test, there will only be a log produced by the core.

gouse045, 2019-12-09:
Hi @udinator,
I am also facing this issue. actually, if run the below command:
python3 run.py --test riscv_arithmetic_basic_test --simulator questa

In the output directory, not getting asm file. because it is written for the riscv_csr_test.

Can u suggest what is the exact flow of the random instruction_generator.

udinator, 2019-12-09:
Hi @gouse045 - if I understand correctly, you are running the riscv_arithmetic_basic_test but are not getting an output assembly test file?

gouse045, 2019-12-10:
Yes @udinator.

udinator, 2019-12-10:
I see. Can you post the compilation log and the simulation log that you are getting output? The command you are running should be correct and should result in the asm files being generated

gouse045, 2019-12-10:
Hi @udinator 
To run the below command:
python3 run.py --test riscv_arithmetic_basic_test --simulator questa

It needs assembly file to run. But the assembly file need to be generated by run.py  In the run.py, it is written for the csr_test.
Please find the compile.log 
[compile.log](https://github.com/lowRISC/ibex/files/3943240/compile.log)
Please suggest me..

udinator, 2019-12-10:
It seems that compilation of the instruction generator is failing for you, which is why the riscv_arithmetic_basic_test is not getting generated; the riscv_csr_test uses a separate python script for generation, which is why you are getting the riscv_csr_test asm file output. This seems to be the main error that is killing your compilation:
`** Error: /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv-dv/src/riscv_instr_pkg.sv(716): Cannot find 'include file riscv_core_setting.sv in directories:
    /home/ubuntu/intelFPGA/16.1/modelsim_ase/verilog_src/uvm-1.2/src, /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv_dv/src, /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv-dv/test, /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv-dv/user_extension, /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv_dv/target/rv32imc, /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv_dv/out_2019_12_06/asm_tests, /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv-dv/src, /home/ubuntu/Desktop/lowrisc/ibex/vendor/google_riscv-dv/test, /home/ubuntu/intelFPGA/16.1/modelsim_ase/verilog_src/uvm-1.2/src`

This is strange, as the command `python3 run.py --test riscv_arithmetic_basic_test --simulator questa` uses the default target of `rv32imc` under `google_riscv-dv/target/rv32imc`. This directory should contain the file `riscv_core_setting.sv`. Can you check if this file `/google_riscv-dv/target/rv32imc/riscv_core_setting.sv` exists?

udinator, 2020-02-04:
Closing this issue since it's been two months. Feel free to open a new issue should anything else come up.

closed 2020-02-04 by udinator