lowRISC/ibex #526: [bitmanip] Implement CLZ instruction

imphil, 2019-12-17:
Implement the CLZ instruction of the bitmanip extension (count leading zeros). This is the first instruction to implement from bitmanip, it will therefore also require a bit of infrastructure work. See the documents linked in #89 for more information.

ganoam, 2020-03-09:
The instructions `clz`, `ctz` and `pcnt` can be implemented to share much of the architecture.

- `clz`: Count Leading Zeros. Counts the number of 0 bits at the MSB end of the argument.
- `ctz`: Count Trailing Zeros. Counts the number of 0 bits at the LSB end of the argument.
- `pcnt`: Counts the number of set bits of the argument.

I propose the following approach:
  - 32 one bit adders, counting the set bits of a signal `bitcnt_bits`
  - For `pcnt` the argument is fed directly into `bitcnt_bits`.
  - For `ctz`: generate a signal with the number of set bists == number of trailing zeros:
    `bitcnt_bits` = (operand - 1) & ~operand;
      - Example:
      ```
      a = 8'b10011000;

      ctz(a):
        10011000
      - 00000001
      ----------
        10010111
      & 01100111
      ----------
        00000111
      ----------
      ```
  - For `clz`: reverse the operand and apply strategy for ctz.

The SV code would look like this:

```
  logic bitcnt_ctz;
  logic bitcnt_pcnt;

  assign bitcnt_ctz  = (operator_i == ALU_CTZ);
  assign bitcnt_pcnt = (operator_i == ALU_PCNT);

  logic [31:0] bitcnt_data;
  logic [31:0] bitcnt_bits;

  assign bitcnt_data = bitcnt_ctz ? operand_a_i : operand_a_rev;
  assign bitcnt_bits = bitcnt_pcnt ? bitcnt_data : (bitcnt_data-1) & ~bitcnt_data;

  logic [5:0] bitcnt_result;
  always_comb begin
    bitcnt_result = '0;
    for (int unsigned i = 0; i < 32; i=i+1) begin
      bitcnt_result[5:0] = bitcnt_result[5:0] + bitcnt_bits[i];
    end
  end
```

davideschiavone, 2020-03-09:
For the operation `(bitcnt_data-1)`, it's worth to try to see if using the existing adder makes sense in terms of area utilization. Your option implies a simple adder (one operand is constant to 1), the second option requires multplexers to selected the operand "-1" (i.e. 0xFFFF_FFFE with carry in equal to 1).

ganoam, 2020-03-09:
Yes. Very good idea.

ganoam, 2020-03-27:
I think this Issue could be closed now.

imphil, 2020-03-27:
Implemented as part of #683

closed 2020-03-27 by imphil