lowRISC/ibex #534: Inconsistency between Spike and OVPSIM

paulzcgu, 2019-12-21:
The post comparison results are different when ISS switched from Spike to OVPSIM for two particular cases:
* riscv_unaligned_load_store_test
* riscv_reset_test

### Environment:
```
ovpsim build from commit: db82ac94ae9d9d2e1182c947424b78e340fd4dae
spike build from commit: master
vcs version: 2019.06-SP2
python: 3.6
```

### Issue:

The results of the `post_compare` step

with ISS = 'spike':
* riscv_unaligned_load_store_test:  Compare result[FAILED]: 84 matched, 18 mismatch
* riscv_reset_test:  Compare result[FAILED]: 0 matched, 22 mismatch

with ISS = 'ovpsim':
* riscv_unaligned_load_store_test:   Compare result[PASSED]: 13699 matched, 0 mismatch
* riscv_reset_test:   Compare result[PASSED]: 0 matched, 0 mismatch

The tests were run on the same set of asm_test files.

The result was persistent with various seeds.

### Note:

The `riscv_unaligned_load_store_test` mismatch may related to issue: 
https://github.com/google/riscv-dv/issues/283#issuecomment-567284887
@taoliug  suggested 
> You can set unaligned load/store to F to see if it solves the problem.
> https://github.com/google/riscv-dv/blob/master/target/rv32imc/riscvOVPsim.ic#L7

### Stimulus:
This zip file includes the .bin, .S and .o files for both tests.
[stimulus.zip](https://github.com/lowRISC/ibex/files/3990943/stimulus.zip)

### Simulation results:
This zip file includes the .log and .csv file for spike, ovpsim and rtl simulation. All the simulations were run with stimulus provided above.
[sim_results.zip](https://github.com/lowRISC/ibex/files/3990951/sim_results.zip)

taoliug, 2019-12-21:
Please note ibex does support unaligned load/store, ovpsim has been configured to match RTL behavior. If you want to simulate with spike, please enable unaligned load/store and build spike again
https://github.com/riscv/riscv-isa-sim/blob/master/config.h.in#L85

paulzcgu, 2019-12-21:
> Please note ibex does support unaligned load/store, ovpsim has been configured to match RTL behavior. If you want to simulate with spike, please enable unaligned load/store and build spike again
> https://github.com/riscv/riscv-isa-sim/blob/master/config.h.in#L85

@taoliug , thanks for the clarification
I didn't change the ibex's testlist.yaml. https://github.com/lowRISC/ibex/blob/e9a6295e57e5ef83ba6fe4d1324a2abbb7032428/dv/uvm/riscv_dv_extension/testlist.yaml#L370
This test was enabled by default.

> ovpsim has been configured to match RTL behavior
Am I correct that OVPSIM is the preferred ISS simulator for ibex?

taoliug, 2019-12-21:
Both spike and ovpsim are supported. ovpsim is more configurable. As I mentioned in the previous comment, in order to get spike to match ibex RTL behavior, you need to rebuild the tool with unaligned load store enabled.

paulzcgu, 2020-01-07:
Tested with a rebuilt spike. All works now. Thanks

closed 2020-01-07 by paulzcgu