lowRISC/ibex #540: Check how X values end up in C decoder

vogelpi, 2020-01-03:
Currently, X values can end up in the C decoder when the IF stage does not have a valid instruction. This requires the assertions in the C decoder to be qualified with `instr_valid` (see #536).

We allow X values for uninitialized memories and the DV environment can drive them on the external bus. It seems that the X can come from both the uninitialized flops in the prefetch buffer (`ibex_fetch_fifo.sv`) or the bus signal directly, or a combination of the two (depending on the requested address, and FIFO fill status).

We need to understand where exactly these X come from in this particular case.

nmoroze, 2020-01-23:
Although I don't think I encountered the exact issue being discussed here, I did notice some uninitialized flops with X values in `ibex_fetch_fifo` (specifically `err_q` and `rdata_q`). I haven't have a chance to trace them and see if it's possible for those particular X's to propagate, but I'm curious if there's any reason not to initialize them on reset just to be safe?

rswarbrick, 2021-04-05:
Hi @vogelpi. Do you know how to trigger these failures? I just did the obvious thing (removing the `valid_i` preconditions on the assertions in the compressed decoder and running some tests) and everything seems to have worked. Do you remember what tests you were seeing these X values from?

vogelpi, 2021-04-06:
Hey @rswarbrick ,
I don't remember how to trigger these failures. I didn't observe them myself back then and I also couldn't find anymore a related issue. Since I opened this issue, a lot of work has been done on the IF side (branch predictor, icache, performance fix to fetch FIFO etc.). For sure also on the DV side there has been a lot of work but there I don't really have a good overview.

Looking into the RTL it seems that `err_q, rdata_q` still don't have a reset (probably because it's not strictly required) and I think that's okay. As for potential sources of X: the valid signal gating the assertion ultimately also comes from the fetch FIFO / bus signal, but it's generation does not involve flops without reset. So we should catch all cases where the X is introduced into the instruction internally. As for X received over the external bus, is it possible that the tests no longer access uninitialized memory?

Anyway, I think it wouldn't hurt to add one more assertion inside the compressed decoder to make sure the `valid` signal itself is not X. If any of the assertions fire, we can debug it. Does that sound reasonable?

rswarbrick, 2021-04-06:
Sounds like a good plan to me.

vogelpi, 2021-04-06:
Okay great! I've filed a PR to add this assertion. Once it is merged, we can close this issue.

closed 2021-04-06 by vogelpi