lowRISC/ibex #548: Illegal instruction assertion fire

udinator, 2020-01-09:
Hi all,
I'm seeing a potential issue where, upon decoding an illegal instruction that contains `x` within it, Ibex will display that the instruction is illegal, but assertions in the LSU will fire, causing the simulation to error out. The simulation log is shown below,
[image]
We can see that at `PC=0x80006232` Ibex recognizes that there is an illegal instruction, and the core does in fact respond correctly by trapping to the proper exception handler, but then the `IbexDataOffsetKnown` assertion in the LSU constantly fires due to an `x` from this instruction propagating through the design, failing the simulation.
What do you think about additionally qualifying these assertions that use `$isunknown` with a check ensuring the instruction is not illegal?
Udi

GregAC, 2020-01-13:
I'll pick this up. I'm making some changes to the decoder as part of my timing optimisations for the branch target ALU so I want to get this test working to validate them.

GregAC, 2020-01-14:
Solving this is a bit of a tricky one, or rather choosing which of the obvious solutions is the way to go is tricky.

The decoder pulls Xs out of the illegal instruction and these become immediate operands/select signals for operands muxes/register addresses etc. This results in Xs being fed into the ALU so Xs in the calculated address which is where data_offset comes from and hence trigger the assertions. 

One could argue that the code should not propagate X so the easy solution is to factor the illegal_insn into all decoder outputs (0 them or force some other default value if we have illegal_insn). However this is not good from a timing perspective. A lot of the decoder outputs are timing critical (especially things like mux select signals for operand selection) and illegal_insn is quite late relative to other decoder outputs. So by factoring in there we will likely make the critical path worse as well as increasing the length of many other paths.

In this particular instance you could factor illegal_insn into data_offset (so it's zeroed on illegal_insn) as data_offset itself comes from the ALU output so factoring in illegal_insn shouldn't cause a big timing impact (though will clearly add a gate delay to the path). However in general there may be other similar situations where we want to avoid fixing it like this because of timing.

You could simply qualify the assertion with illegal_insn

```verilog
`ASSERT(IbexDataOffsetKnown, ~illegal_insn |-> data_offset, clk_i, !rst_ni)
```

Though from the style guide it seems we were aiming for a straight-forward inputs into a case statement must not be unknown style which this conflicts with.

Finally we could say valid instruction data generated by the DV environment must not have Xs in (i.e. it's all X and instr_rvalid_i isn't set or instr_rvalid_i is set and every bit is not X)

I'd tend towards the third option (modify assertion to stop it firing in this case) though think we should write some stuff in the style guide about this. I'll open an issue in the style guide to discuss further.

GregAC, 2020-01-14:
@imphil @udinator @vogelpi @tomroberts-lowrisc thoughts appreciated.

tomroberts-lowrisc, 2020-01-14:
I'd tend towards 3 or 4, probably 4.
I view X insertion as more of a signal protocol-level check, i.e. the rdata should be X when rvalid isn't high, but not otherwise. If rvalid is high the actual instruction word can be garbage, but not X. Most protocol checkers would enforce that on typical bus protocols.

vogelpi, 2020-01-14:
Thanks @GregAC for looking into this.

IMO Option 3 is the best but I think it is problematic that `illegal_insn` is masking assertions even if the instruction is known/has no X in it. Question: What is the desired behavior in the case of an illegal instruction with X in it? Should the simulation abort because of an SVA firing in the decoder or do we expect the simulation to continue running and the core to handle the illegal instruction exception?

GregAC, 2020-01-14:
Well my suggestion for the style guide was going to be along the lines of if you want to qualify the !$isunknown assertion for a case input you must have unqualified ASSERT_KNOWN on the case statement outputs (or alternatively say you have unqualified ASSERT_KNOWN on the inputs or the outputs, with input preferred). So we wouldn't end up totally disabling assertions around this case statement for illegal instructions.

As for desired behavior of an illegal instruction with X in, if that instruction would be illegal for all possible values of the bits that are X then I say it should trigger an illegal instruction exception and no assertion should fire.

Though if we say no Xs on instr_rdata if we have instr_rvalid then we don't need to worry about this.

GregAC, 2020-01-14:
Opened the issue against the style guide (see github's helpfully generated auto-link). Realised that ASSERT_KNOWN on the output is probably pointless given case statements that follow our style guide cannot generate X anyway.

PR #591 (2020-02-10): [rtl] Fix assertion issues
Fixes #548

Fixes the long standing assertion problem from the linked issue as well as a new problem that was introduced with the branch target ALU.
commit 24cbc322 (2020-02-10): [rtl] Fix assertion issues

Fixes #548
closed 2020-02-10 by GregAC