lowRISC/ibex #557: Nested NMI interrupts

udinator, 2020-01-16:
Hi,
When debugging a test case in which I asserted two nested NMI interrupts to force the core to respond to the second NMI interrupt while executing the handler for the first NMI, resulting in this waveform:
[image]
At the location of the cursor, Ibex is in the middle of executing the NMI interrupt handler and sees the input interrupt lines being changed to another set of incoming interrupts; these new interrupts are processed when `csr_save_cause_i` goes high at `3,843,000ns`.  Note that `nmi_mode_q` in the controller is high before the second set of interrupts due to the NMI detection.
However, this second set of interrupts also includes a NMI, but Ibex sets `mcause` to `0b11_1110 = 0x3e` indicating a fast interrupt, causing a testbench failure since we should expect `mcause` to be `0b11_1111 = 0x3f` for the NMI. This mismatch is due to the `IRQ_TAKEN` state in the controller, specifically this section:
[image].
Since `nmi_mode_q` is already high from the first NMI that Ibex was initially handling, the core does not recognize the NMI asserted inside of the interrupt handler, and as such delegates it to be a fast interrupt, causing a mismatch in `mcause`.

Another potential issue that could arise is from this section in the CSR register file:
[image]
Since Ibex restores CSR state from the `mstack` CSRs only in the case of NMI interrupts, I don't think the core would be able to correctly return from a series of nested NMIs, as the saved state from the first NMI in the `mstack` CSRs would be overwritten by the state of the second NMI.

Given these potential issues, my question is if this situation of nested NMI interrupts is something that Ibex can handle at all?

vogelpi, 2020-01-17:
Hi @udinator,
thanks for filing this issue and the detailed report.

Let me answer your last question first: No, Ibex cannot handle nested NMIs. There is only one NMI line and when Ibex is in NMI mode already (`nmi_mode_q`) any activity on the NMI line is ignored. Activity on other interrupt lines must be ignored by not setting mstatus.MIE back to 1 (this would need to happen in software, the hardware sets this to 0 automatically when taking an interrupt).

Regarding the problem that the controller takes the wrong interrupt at `3,843,000ns`, this has the following meaning: mstatus.MIE is set to 1 by software in the test which is not expected to happen. We could also mask this in hardware when `nmi_mode_q` is set but I don't think this is needed as software really should not enable interrupts when handling an NMI.

To summarize:
- Ibex has only one NMI line and cannot handle multiple nested NMIs. Once in NMI mode, activity on the NMI line is ignored.
- As with any other interrupt, the handler should clear the NMI at the source at some point after entering the handler. Ibex will only look at new activity on this line when leaving NMI mode.
- When handling any interrupt, other interrupts should only ever be activated (by setting mstatus.MIE backt to 1) when the core is indeed supposed to handle them.

imphil, 2020-01-17:
Thanks for your answer @vogelpi. I think we had now two people being confused by that. Is there anything we can do to improve (the visibility of) the documentation, either in the source code, and/or in the rst documentation?

GregAC, 2020-01-17:
Is it actually specified anywhere in the RISC-V specs that when you're handling an NMI you ignore the NMI line?

Whilst we don't support nested NMIs I'd say the current behaviour of being in an NMI handler, setting mstatus.MIE, seeing another NMI and ending up with an mcause indicating a fast interrupt is surprising.

From my quick reading of the specs when we get an NMI we should always jump to the NMI handler regardless of whether we're in 'NMI mode' or not and xIE bits shouldn't effect anything.

udinator, 2020-01-17:
@vogelpi - Thanks for the detailed answer! I am indeed setting mstatus.MIE back to 1 in the interrupt handler for these nested interrupt scenarios, just like you said. Correct me if I'm wrong, but from what you are saying, it sounds like nested interrupts should not be allowed during NMI handling. Would it be possible to either mask this off in hardware or specify this in the documentation? Like @GregAC mentioned, the spec is rather vague about how NMIs are dealt with, so I found myself pretty surprised upon seeing a nested NMI being treated like a fast interrupt.
I'll also go ahead and modify our testbench to not allow nested interrupts during handling of an NMI to more closely align to the behavior that you specified.

udinator, 2020-01-17:
A follow-on question somewhat related to this discussion: do we allow debug requests to be sent to the core during handling of NMI interrupts?

vogelpi, 2020-01-29:
Sorry guys for my late response. I completely forgot about this issue. Let me answer your questions.

@GregAC :
> Is it actually specified anywhere in the RISC-V specs that when you're handling an NMI you ignore the NMI line?

No. As you noted, the spec just says that when we get an NMI we always jump to the NMI handler. But I think this does not make sense with level-sensitive interrupts (all interrupts in Ibex are level sensitive) as it would require the core to constantly jump into the handler until the NMI is magically de-asserted. We could make the NMI edge-sensitive to avoid this but I am not convinced this makes sense.

> Whilst we don't support nested NMIs I'd say the current behaviour of being in an NMI handler, setting mstatus.MIE, seeing another NMI and ending up with an mcause indicating a fast interrupt is surprising.

I agree that this is surpising. Maybe the most reasonable thing to here is to just ignore all regular interrupts in NMI mode as well. I will do a PR for that and also adjust the documentation accordingly as suggested by @imphil .

@udinator :
> Do we allow debug requests to be sent to the core during handling of NMI interrupts?

Debug requests are always enabled unless we are in debug mode already. The spec does not say anything about ignoring debug requests while handling an NMI. I think we should not change that.

udinator, 2020-01-29:
@vogelpi - I agree with the reasoning that all interrupts should be completely masked off while handling an NMI, I think it would best align with the mechanism that Ibex implements.

sjgitty, 2020-01-29:
> I agree with the reasoning that all interrupts should be completely
masked off while handling an NMI

Would we expect the HW to do this masking? Or the NMI interrupt handler?
Seems like the
former. For this and all nested interrupt cases (how many are handled by
Ibex? What does
the core do if this number is exceeded?) let's make sure the documentation
is clear and the
DV follows what the document says (and tests it for correctness of course).

vogelpi, 2020-01-30:
> Would we expect the HW to do this masking? Or the NMI interrupt handler? Seems like the former.

Yep, that is how the linked PR modifies the design: All interrupts are masked while in NMI mode.
> For this and all nested interrupt cases (how many are handled by Ibex? What does the core do if this number is exceeded?) let's make sure the documentation is clear and the DV follows what the document says (and tests it for correctness of course).

The linked PR clarifies the doc to state that nesting is supported in software (interrupts need to be re-enabled after saving `mepc`, `mstatus` in software) but that there is not support for hardware nesting, and that the `mstack` CSRs are for recoverable NMIs only.

vogelpi, 2020-02-11:
I am closing this issue the corresponding PR modifying the RTL and clarifying the documentation got merged a while ago. Feel free to re-open if necessary.

closed 2020-02-11 by vogelpi