lowRISC/ibex #562: failed in running make

nak-tech, 2020-01-23:
the hierarchy of riscv compiler is 
/opt/riscv/
|--bin
|--include
|--lib
|--libexec
|--riscv32-unknown-elf
|--riscv64-unknown-linux-gnu
|--share
|--sysroot

in example/sw/led directory, run command:  make CC=/opt/riscv
log is:

/opt/riscv -march=rv32imc -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall -g -Os -MMD -c  -o led.o led.c
make: execvp: /opt/riscv: Permission denied
Makefile:50: recipe for target 'led.o' failed
make: *** [led.o] Error 127

Any advice? thanks.

GregAC, 2020-01-23:
The CC parameter should point to a compiler binary, not to the root of the tree containing the toolchain.

So in your case

```
make CC=/opt/riscv/bin/riscv32-unknown-elf-gcc
```

May do the trick

nak-tech, 2020-01-23:
I do not have -gcc path in my /opt/riscv folder. tried following 2 options:

 make CC=/opt/riscv/riscv32-unkonwn-elf-gcc
/opt/riscv/riscv32-unkonwn-elf-gcc -march=rv32imc -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall -g -Os -MMD -c  -o led.o led.c
make: /opt/riscv/riscv32-unkonwn-elf-gcc: Command not found
Makefile:50: recipe for target 'led.o' failed
make: *** [led.o] Error 127

 make CC=/opt/riscv/riscv32-unkonwn-elf
/opt/riscv/riscv32-unkonwn-elf -march=rv32imc -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall -g -Os -MMD -c  -o led.o led.c
make: /opt/riscv/riscv32-unkonwn-elf: Command not found
Makefile:50: recipe for target 'led.o' failed
make: *** [led.o] Error 127

imphil, 2020-01-23:
GCC is in the `bin` folder, please use the path exactly as @GregAC has provided it.

nak-tech, 2020-01-23:
stuck at next step:
ghlab@build_riscv_pc:~/work/ibex/examples/fpga/artya7-100$ fusesoc --cores-root=. run --target=synth --setup --build lowrisc:ibex:top_artya7_100

ERROR: Conflicting requirements:
Requirements: 'lowrisc_ibex_top_artya7_100 == 0.1-0' <- 'lowrisc_ibex_fpga_xilinx_shared >= 0-0'
    +lowrisc_ibex_top_artya7_100-0.1-0 was ignored because it depends on missing packages
Requirements: 'lowrisc_ibex_top_artya7_100 == 0.1-0'
    Install command rule (+lowrisc_ibex_top_artya7_100-0.1-0)

Failed to resolve dependencies for lowrisc:ibex:top_artya7_100

imphil, 2020-01-23:
Please run fusesoc in the top-level directory of the source tree, i.e. in `~/work/ibex`.

nak-tech, 2020-01-23:
many thanks. all running.

closed 2020-01-23 by nak-tech