lowRISC/ibex #567: Reads of `mip` CSR should not depend on corresponding fields in `mie`

pfmooney, 2020-01-24:
Currently, the bits in `mip` are ANDed with those in `mie`:
https://github.com/lowRISC/ibex/blob/43752a6c193893d4750daceff5f284f22da52f77/rtl/ibex_cs_registers.sv#L241-L245
This is at odds with the specification, which states the mechanisms by which those bits would be cleared:
-  The MTIP bit is read-only and is cleared by writing to the memory-mapped machine-mode timer compare register
-  MEIP is set and cleared by a platform-specific interrupt controller

And while it does state that "An interrupt _i_ will be taken if bit _i_ is set in both `mip` and `mie`, and if interrupts are globally enabled.", it makes no mention of the asserted bit(s) in `mip` depending on their `mie` counterpart(s).

Looking for other interpretations of this issue, I found that qemu-riscv treats `mip` similarly, where `mie` used to determine if CPU execution should be interrupted, but not whether bits are asserted in `mip`.  A survey of some other open-source RISC-V cores ([e200](https://github.com/SI-RISCV/e200_opensource), [scr1](https://github.com/syntacore/scr1)) yielded the same result: reads of `mip` yielding results not ANDed through `mie`.

vogelpi, 2020-01-27:
Thanks @pfmooney for opening this issue. I think you are right, `mip` should not depend on `mie` but the two are combined to decide whether an interrupt is actually taken. This should be fixed. Hopefully I can do it this week.

I am also including @davideschiavone , I think RI5CY has the same issue.

vogelpi, 2020-01-27:
I am also adding @udinator and @taoliug as this needs to be accounted for in verification as well.

davideschiavone, 2020-01-28:
Thanks @vogelpi  - CLINT porting on RI5CY is actually happening right now so thanks a lot for the tip!

closed 2020-02-04 by vogelpi

davideschiavone, 2020-02-19:
Dear all, Arjan from SiLabs actually noticed that this PR may be wrong.
@vogelpi , can you partecipated at the discussion here https://github.com/openhwgroup/cv32e40p/pull/231 ?

Thanks a lot

GregAC, 2020-02-19:
I think we are compliant with the bit of the spec Arjan highlighted.

Both the global core clock enable the wake from WFI in the controller depend upon irq_pending:

https://github.com/lowRISC/ibex/blob/f90faa6ca365e02047d456ac3ad084ecc457ada0/rtl/ibex_core.sv#L285

https://github.com/lowRISC/ibex/blob/f90faa6ca365e02047d456ac3ad084ecc457ada0/rtl/ibex_controller.sv#L310-L325

irq_pending comes from cs_registers and factors the individual enable bits into the interrupts but not the global enable:

https://github.com/lowRISC/ibex/blob/f90faa6ca365e02047d456ac3ad084ecc457ada0/rtl/ibex_cs_registers.sv#L635-L636

vogelpi, 2020-02-19:
Thanks @GregAC for extracting the relevant code bits :+1: