lowRISC/ibex #569: Add missing casts to enums (most likely in DV code only)

imphil, 2020-01-28:
@danielmlynek discovered this in https://github.com/lowRISC/ibex/pull/566:

All line where illegal assignment are done is now marked with warning ie:
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 781  1

This argument will be no longer needed when explicit cast will be used on such illegal assignment ie in file :
 /home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv, line 201
` operating_mode = core_init_mstatus[12:11];`
need to be changed to:
` operating_mode = priv_lvl_e'(core_init_mstatus[12:11]);`

@danielmlynek do you know of more cases where such a cast is missing?

@msfschaffner I suppose currently the DV code isn't running through Lint, which should have caught that. Can you think of an easy way to lint that code as well? (Either here, or in OpenTitan.)

danielmlynek, 2020-01-28:
Riviera compiler returns this warning for below lines - so all of them need to be reviewed to see if needs explicit casting
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 201  1
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 781  1
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 88  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 91  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 94  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 97  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 100  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 103  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 106  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 109  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 112  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 115  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 118  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 121  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 124  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 127  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 130  32
WARNING VCP2694 "Assignment to enum variable from expression of different type." "/home/danielm/gitlab/ibex/dv/uvm/tests/core_ibex_test_lib.sv" 133  32

imphil, 2020-01-28:
Thanks for the list @danielmlynek. @udinator would you mind having a look at these warnings/errors and add an explicit cast? (Probably you can get VCS to report similar warnings.)

udinator, 2020-01-28:
@danielmlynek - thank you for finding out all of these instances!
@imphil - Sure, I will go through the DV code and scrub all of these warnings.

udinator, 2020-01-28:
@danielmlynek, I've submitted a PR that hopefully address most/all of these Riviera compile warnings. Could you re-run the tool and let me know if anything else comes up?

imphil, 2020-01-29:
PR #572 is now merged, master should run without warnings. Please re-open this issue if some of the issues remain.

danielmlynek, 2020-01-29:
It should be REOPEN - there is one more line in /ibex/dv/uvm/tests/core_ibex_test_lib.sv
mem_status = signature_data_q.pop_front();
should be
mem_status =  core_status_t'(signature_data_q.pop_front());

imphil, 2020-01-29:
Thanks @danielmlynek! 
@udinator can you have another look? (Or if you like, you can also do a PR to fix this one @danielmlynek.)

udinator, 2020-01-29:
@danielmlynek thanks!
I've opened PR #576 to fix this last issue.

msfschaffner, 2020-02-10:
Thanks for cleaning this up (and sorry for being late to the party here - I was out the last two weeks). Indeed, we only run RTL through AscentLint at the moment.

udinator, 2020-02-29:
@danielmlynek - are you still seeing these issues?

danielmlynek, 2020-03-02:
No. 
Now all looks ok reagrding this issue.

imphil, 2020-03-02:
Great, thanks for confirming @danielmlynek. Closing this then.

closed 2020-03-02 by imphil