lowRISC/ibex #574: Fetch FIFO blocks instruction request when full.

ganoam, 2020-01-29:
Hi everyone, 

This is the first time I raise an issue, so please let me know if I am too verbose, not verbose enough, incoherent, or generally out of place. 

I may have found an issue with the prefetch buffer. The situation: I have implemented a second memory port in the simple system example (PR #571) and run both the hello_test example and coremark on it. Since I would expect no memory contention on the instruction line, i would expect to record less IFetch waits with coremark. however the opposite was the case: Using the separated data and instruction ports, coremark recoreded around 200 more IFetch wait than using a single memory port.
For the hello_test example, I have found that the new implementation takes two more cycles than the baseline. By comparing the traces of the two hello_test runs, I think I can pinpoint exactly where the problem is.

The top level instruction request signal `instr_req_o` is set in `prefetch_buffer.sv` lines 108-111 by the signal `valid_req`:  
``` 
// Make a new request any time there is space in the FIFO, and space in the request queue
assign valid_new_req = req_i & (fifo_ready | branch_i) & ~rdata_outstanding_q[NUM_REQS-1];
assign valid_req = valid_req_q | valid_new_req;
```
So if we re not branching, a memory request can only be issued when the fifo is ready. The fifo ready signal is set in `ibex_fetch_fifo.sv` line 155:
```
assign in_ready_o = ~valid_q[DEPTH-NUM_REQS]; // DEPTH = 3, NUM_REQS=2
```
Now, by comparing the traces I have found that the delay is introduced around 154ns into the hello_test simulation. The waveform for the baseline implementation is here:
[image]

The waveform for the implementation with dual port memory is here:
[image]

At 143ns you can see that the pipeline is stalled due to the `id_stage` being busy. This is a store-word instruction, so it makes a data request. In the baseline implementation the data and instruction line is shared, so in the meantime no instruction can be fetched. However for the dual-port implementation a new instruction is fetched, and the fetch-fifo after that has stored two memory words. therefore the fifo ready signal will be low, and no new memory request will be made (`instr_req_o`).  
The first instruction (`PC=001003B4`) in the FIFO is compressed. So, after that the word can not be popped and the fifo is still full. Here is content of the fifo at that point (at 149ns):
```
             | 31             16 | 15              0 |
FIFO entry 0 | PC001003B6 [15:0] | PC001003B4 [15:0] |
FIFO entry 1 | PC001003BA [15:0] | PC001003B6 [31:16]| 
```
So instruction `001003BA` is fetched and in the next cycle the fifo is not full anymore. 
```
             | 31             16 | 15              0 |
FIFO entry 0 | PC001003BA [15:0] |                  | 
FIFO entry 0 |                   |                   |
```
So  `instr_req_o` is raised again (at 151ns), but `instr_rvalid_i` is only received in the cycle after that, leading to a 1 cycle stall, since the instruction in the fifo is incomplete.

For the baseline implementation the fifo does not get full and therefore `instr_req_o` is never lowered and we never have to wait for `instr_rvalid_i`. That's the reason why the single-port implementation finishes faster.

So, as far as I can tell, at least while there are unaligned, uncompressed instructions in the fetch fifo, it can never remain full. Thus there is always a stall while waiting for `instr_rvalid_i`. 

I think the fundamental problem is that no simultaneous pop and push is possible while the FIFO is full. If we would take into account the anticipated next-cycle occupancy when setting the fifo `in_ready_o` signal, i. e. change line 155 `ibex_fetch_fifo.sv` to
```
assign in_ready_o = ~valid_d[DEPTH-NUM_REQS]; 
```
the memory request could be done one cycle earlier and the valid signal will arrive only once there is space in the fetch FIFO. 

I have run the hello_test and coremark programs using this fix. The number of IFetch waits for coremark using the fix on the baseline (single-port) implementation decreases from 14968 to 5215. Total ticks decrease from 423620 to 416639.  

Executing the hello_test program, there is no improvement for the single-port implementation. However the additional two cycles of latency that were observed for the dual-port implementation disappear. 

Also coremark now executes with exactly the same statistics for single- and dual-port implementation. This is disapointing, but I think very well possible for non-data intensive workloads. 

I do not know if my fix is fully correct, since there are some parts in the Prefetch Buffer and the Fetch FIFO, that I don't yet completely understand. But I hope that my account may help somebody investigate and fix the problem.

I will attatch the output of my simulation runs. For coremark I will omit the trace_core log and waveforms, since those are huge files. If you are interested in them let me know.
[prefetch-issue_sim-out.zip](https://github.com/lowRISC/ibex/files/4127274/prefetch-issue_sim-out.zip)

Nomenclature of the folders: master = single-port implementation, perf-eval = dual-port implementation

In perf-eval_hello_test-out, there is a gtkwave save-file `waveconfig.gtkw which you can use with all the simulation waveforms to directly display the waveforms as they are in my screenshots.

cheers, Noam

tomroberts-lowrisc, 2020-01-29:
Thanks for the detailed report, and all the investigation that has gone into it.
You are correct that the prefetch buffer / fetch FIFO has some inefficiencies which lead to stalls that could be avoided.
Can you elaborate further on your fix? I have a rough version of the RTL where I fixed this issue by counting free entries + outstanding requests to determine if a new fetch is permitted, but it is not ready for PR yet.

ganoam, 2020-01-29:
Yes I will try - I think I attached the wrong files for the simulation waveforms of
 the 'fixed' implementations.  I attach it here again propperly.
[prefetch-buff-issue_sim-out_new.zip](https://github.com/lowRISC/ibex/files/4127653/prefetch-buff-issue_sim-out_new.zip)

So, at time 147ns in the dual-port simulation in the above screenshot, `valid_q==3'b011`. This is the condition to reset the fifo `in_ready_o` signal in `ibex_fetch_fifo.sv`, since at that point the fifo is full. The instruction memory-request depends on that signal: a memory request is only issued when the fifo is ready. Once a memory request is issued, it is immediately granted in my new dual-port implementation (since it has its own lines) by raising `instr_gnt_i`.
However the instruction data is valid (`instr_rvalid_i`) only in the next cycle.

So, the ready signal and thus the memory request is set once the fifo is not full anymore. But the data arrives only one cycle later. So the prefetch buffer sits there with one valid word - if it is an aligned instruction or a compressed instruction it can feed it to the ID stage. But if it is an unaligned uncompressed instruction, only half the instruction data is there and we have to wait for the second half. 

Now, my 'fix' sets the fifo ready signal in anticipation of the delay of the arrival of data: It is ready, if there will be space in the next cycle. This is done by making the ready signal dependant on not the current state, but the next state of the FIFO. It should therefore be able to simultaneously push and pop also when full, which I think, is not possible with the baseline implementation.

Here is the screenshot of the fixed dual-port implementation waveform (which was not in the original zip file - sorry.)
[image]

You can see how the instr_req_o only goes low for one cycle at 145ns. during the following cycle the fifo becomes full, because only a compressed instruction is fetched. During that cycle, `valid_d[2:0] ==001` indicates that the fifo is popping a word, therefore there will be space again in the next cycle. So the `instr_req_o` signal is set again in order to receive the next instruction in the subsequent cycle.

GregAC, 2020-01-29:
I believe I also came across this issue whilst evaluating the performance of the third pipeline stage (but had yet to file an issue for it or look into it in great detail). I noted instances where we were stalled awaiting an instruction that could have been avoided as the prefetch buffer wasn't full and hadn't been trying any new requests in prior cycles.

A quick workaround was simply making the prefetch buffer bigger (I think I arbitrarily chose DEPTH = NUM_REQS + 4). However a proper fix along the lines you propose is definitely called for.

PR #907 (2020-06-01): [rtl] prefetch buffer performance fix
- The prefetch buffer needs to know when space is available in the fetch
  FIFO to accept a new external request.
- This change updates that logic to look at what is in the FIFO and what
  is outstanding on the bus to decide when space is available rather
  than always assuming the maximum number of requests are outstanding.
- This improves the usage efficiency of the FIFO and fixes #574
commit 4b01580a (2020-06-01): [rtl] prefetch buffer performance fix

- The prefetch buffer needs to know when space is available in the fetch
  FIFO to accept a new external request.
- This change updates that logic to look at what is in the FIFO and what
  is outstanding on the bus to decide when space is available rather
  than always assuming the maximum number of requests are outstanding.
- This improves the usage efficiency of the FIFO and fixes #574
closed 2020-06-01 by tomroberts-lowrisc