lowRISC/ibex #578: Update Ibex Simple System readme for Riviera-PRO

dawidzim, 2020-01-31:
Hey!
Can you update Ibex Simple System readme for Riviera-PRO?
To build and run this example on Riviera-PRO we need to use command:

fusesoc --cores-root=. run --target=sim --tool=rivierapro lowrisc:ibex:ibex_simple_system --RV32M=1 --RV32E=0 --SRAM_INIT_FILE=../../../examples/sw/simple_system/hello_test/hello_test.vmem --vsim_options=+ibex_tracer_file_base=test_results

--SRAM_INIT_FILE - path to hello_world binary
--vsim_options=+ibex_tracer_file_base=output_file_name - here we can change the output file name

imphil, 2020-01-31:
Of course. Do you feel comfortable doing a pull request with your proposed changes? Or otherwise just me know and we'll do it.

dawidzim, 2020-01-31:
I would be grateful if you did this

commit 3f0b730d (2020-02-05): [doc] Riviera-PRO instructions for Simple System

Fixes #578
closed 2020-02-05 by GregAC