lowRISC/ibex #579: Use Verible for lint

imphil, 2020-02-01:
We want to use verible_lint (https://github.com/google/verible) as one of our lint tools. There are a couple things to be solved first.

A run of Verible today shows that most lint rules match our use of SV, with the exception of localparam naming (which verible_lint expects to be UpperCamelCase, while we go for ALL_CAPS).

- [ ] Extend edalize to support verible: https://github.com/olofk/edalize/issues/95
- [ ] Add Verible as a lint target to our .core files

If we want to run Verible in "enforcing" mode (e.g. in CI) we additionally need:
- [ ] Waiver support: https://github.com/google/verible/issues/105

CC @fangism @msfschaffner @mithro

msfschaffner, 2020-02-10:
The localparam discrepancy is somewhat unfortunate. Initially, the lowRISC style guide was based on the one that Verible uses, but it looks like we have diverged here. Currently, the lowRISC style makes the distinction depending on whether the parameter/localparam is to be interpreted as an immutable constant or not. The style that Verible follows makes the distinction based on whether it is a localparam or a parameter. One way would be to align the style here, but the optimum would of course be configuratbility...

imphil, 2020-02-10:
@msfschaffner You can find the full motivating discussion on the current localparam style guide at https://github.com/lowRISC/stwg-base/issues/427 (Sorry for all non-OpenTitan readers, this issue isn't public. I can copy over content if you're really interested.)

imphil, 2020-07-02:
A couple more things on the Verible front to be fixed, but none of them are blockers:

- https://github.com/google/verible/issues/343
- https://github.com/google/verible/issues/341
- https://github.com/google/verible/issues/344

https://github.com/olofk/edalize/pull/178 is the enablement on the edalize front, and #999 are the remaining fixes on Ibex.

PR #1000 (2020-07-03): Enforce Verible Lint in CI
Add the last remaining things to enforce Verible lint in CI:
- Add a waiver file. The waiver files currently don't support comments, and require the "empty" regex; bugs have been filed to get that resolved upstream.
- Enable it per config
- Add lint for the simple system
- Some core file cleanups.

Fixes #579
closed 2020-07-03 by imphil